Ircama/epson_print_conf #12: Can i help you adding ET-4700 ?

DonKiShoot, 2024-08-30:
Hello,

I have the issue where waste ink block the printer.
I tried your tool but apparently this doesn't work.

[INFO] Detecting printers... (this might take a while)
[ERROR] 'utf-8' codec can't decode byte 0x88 in position 18: invalid start byte

Version windows 2.1.28

Regards,

Ircama, 2024-08-30:
Please try now v2.1.30

DonKiShoot, 2024-08-30:
Detection seems ok now : 

[INFO] Detecting printers... (this might take a while)
[INFO] Found printer 'EPSON ET-4700 Series' at 192.168.1.17 (hostname: epson8177c2.home)

I will try the others functions :)

Thank for your rapidity ;-)

DonKiShoot, 2024-08-30:
"Power-off timer get" gave : 

[ERROR] Please select a printer model and a valid IP address, or press 'Detect Printers'.

DonKiShoot, 2024-08-30:
Same for "TI Received Time" : 

[ERROR] Please select a printer model and a valid IP address, or press 'Detect Printers'.

DonKiShoot, 2024-08-30:
Printer status seems working (i didn't capture all voluntarily)

[image]

DonKiShoot, 2024-08-30:
[image]

[image]

Only "IP Address" and "Power Off Timer" values are missing (it s empty)

DonKiShoot, 2024-08-30:
"Reset Waste Lnk Levels" seems not working :-(

[INFO] Detecting printers... (this might take a while)
[INFO] Found printer 'EPSON ET-4700 Series' at 192.168.1.17 (hostname: epson8177c2.home)
[ERROR] Please select a printer model and a valid IP address, or press 'Detect Printers'.
[ERROR] Please select a printer model and a valid IP address, or press 'Detect Printers'.
[ERROR] Please select a printer model and a valid IP address, or press 'Detect Printers'.

commit a64fea1e (2024-08-30): Add ET-4700

Resolves #12
closed 2024-08-30 by Ircama