Ircama/epson_print_conf #15: Ink PAD Counter Reset Not Happening

vkabcd, 2024-09-05:
The app is able to find the printer, and reset worked as per message, but still printer boot with message "printer ink pad is end of service life", any suggestions to reset.

[INFO] Detecting printers... (this might take a while)
[INFO] Found printer 'EPSON XP-820 Series' at 192.168.1.147 (hostname: EPSON09BE09)
[INFO] Waste ink levels have been reset. Perform a power cycle of the printer now.
[ERROR]: Missing 'Power off timer' in configuration
[ERROR]: Missing 'First TI received time' in configuration
[INFO] Waste ink levels have been reset. Perform a power cycle of the printer now.
[INFO] Waste ink levels have been reset. Perform a power cycle of the printer now.

Ircama, 2024-09-05:
Please try now

vkabcd, 2024-09-05:
Great... It Worked Flawlessly... Thank you,

closed 2024-09-05 by vkabcd