Ircama/epson_print_conf #19: Borderless ink pad Epson ET-16650

Virusmater, 2024-09-14:
I have following error with my printer:
```
You cannot print borderless until the printer's borderless printing ink pad is replaced. The ink pad is not user-replaceable. Please contact Epson support.

You can still print when not using the borderless printing feature.
```
Printer is detected, but not supported:
```
[INFO] Found printer 'EPSON ET-16650 Series' at 192.168.1.100 (hostname: EPSON794D1C)
```
[image]
[image]
[image]
[image]
[image]

Any idea if I can reset **borderless** printing ink pad? I tried to run using different printer model and do the reset, but it didn't work

Ircama, 2024-09-19:
Printer ET-16650 is not defined in [epson_print_conf.py](/Ircama/epson_print_conf/blob/main/epson_print_conf.py).

Have you tried using [parse_devices.py](https://github.com/Ircama/epson_print_conf?tab=readme-ov-file#parse_devicespy)?

Virusmater, 2024-09-19:
Sorry, I cannot check anymore, I did the reset with Wic

closed 2024-09-19 by Virusmater