Ircama/epson_print_conf #21: when trying to reset the level of used ink, it gives an error

LAYZ1HAG, 2024-09-18:
[image]
Please help me, what am I doing wrong?

Ircama, 2024-09-19:
Please try with version 2.2.1

LAYZ1HAG, 2024-09-19:
for some reason, the new version of the program did not run through the exe, I was able to run it using python and got the same error
[image]

Ircama, 2024-09-20:
Please try again with version [v2.2.11](https://github.com/Ircama/epson_print_conf/releases)

LAYZ1HAG, 2024-09-20:
ok, now the program has started with the exe, and there are explanations for the errors
[image]

Ircama, 2024-09-20:
This issue should be fixed with v2.2.14.

closed 2024-09-20 by Ircama

LAYZ1HAG, 2024-09-20:
Thanks bro, it's working

commit aeca76ef (2024-09-20): Fix "write_key" error for printer L366

Resolve #21