Ircama/epson_print_conf #22: failed to launch GUI

fredfred41, 2024-09-19:
Hi,

Using Win11 and trying to reset Epson ET-2711, I launched the exe and I have this error, could you tell me how to fix it, please?
Thanks a lot

```
Traceback (most recent call last):
  File "gui.py", line 3, in <module>
  File "PyInstaller\loader\pyimod02_importers.py", line 378, in exec_module
  File "ui.py", line 16, in <module>
ModuleNotFoundError: No module named 'black'
[PYI-37992:ERROR] Failed to execute script 'gui' due to unhandled exception!
```

Ircama, 2024-09-20:
I should have fixed the bug. Printer Epson ET-2711 is not included in the configuration. Please [try integrating it yourself](https://github.com/Ircama/epson_print_conf/?tab=readme-ov-file#example-to-integrate-new-printers).

closed 2024-09-20 by Ircama

fredfred41, 2024-09-20:
Thanks a lot. I don't think I'm capable of doing that by myself.
Will try...

commit 2b77c806 (2024-09-20): Add black to requirements and improve UI

fix #22