Ircama/epson_print_conf #26: Add new printer on ET series

MWMatthieuBR, 2024-10-02:
Hello,
First of all thank you for your program, you saved one of my 2 printers !!!
I followed your instructions to add a new printer with the link on Codeberg, because my first printer is an ET-2756 but the second is an ET-2856 and it is not in your program.

The command only gives me ET-2851 ET-2853 ET-2855 for the 2850 series, I tried to reset the counter with that but it doesn't work, when I restart my printer I get error E-11.

Can you help me to understand why ? Thank you !

MWMatthieuBR, 2024-10-03:
Hello, i view your commit and git reset to have it.
The config given by codeberg is :
````JSON
 "ET-2850": {
            "alias": ["ET-2853", "ET-2855", "ET-2851"],
            "read_key": [74, 54],
            "write_key": b'Arantifo',
            "main_waste": {"oids": [48, 49, 47], "divider": 109.13},
            "borderless_waste": {"oids": [50, 51, 47], "divider": 16.31},
            "third_waste": {"oids": [252, 253, 254], "divider": 15.3},
            "raw_waste_reset": {
                48: 0, 49: 0, 47: 0, 52: 0, 53: 0, 54: 94, 50: 0,
                51: 0, 55: 94, 28: 0, 252: 0, 253: 0, 254: 0, 255: 94
            },
            "stats": {
                "Total print pass counter": [133, 132, 131, 130],
                "Total print page counter - rear": [755, 754, 753, 752],
                "Total print page counter - duplex": [763, 762, 761, 760],
                "Total scan counter": [1843, 1842, 1841, 1840],
                "Ink replacement counter %-- Black": [554],
                "Ink replacement counter %-- Cyan": [555],
                "Ink replacement counter %-- Magenta": [556],
                "Ink replacement counter %-- Yellow": [557]
            },
            "serial_number": range(1604, 1614),
        }
     
 ````
 But with the new code I get an error because he received none for the values :
    [ERROR] int() can't convert non-string with explicit base
Traceback (most recent call last):
  File "C:\Users\***\Desktop\epson_print_conf\ui.py", line 678, in get_current_eeprom_values
    org_values = ', '.join(
                 
  File "C:\Users\***\Desktop\epson_print_conf\ui.py", line 679, in <genexpr>
    f"{k}: {int(v, 16)}" for k, v in zip(
          
TypeError: int() can't convert non-string with explicit base
````

[NOTE] Current EEPROM values for Raw waste reset: 48: None.
[NOTE] Current EEPROM values for Main waste reset: 48: None.
[NOTE] Current EEPROM values for Borderless waste reset: 50: None.

Ircama, 2024-10-05:
> But with the new code I get an error because he received none for the values :

This is because you are using a wrong printer model, with a different "read_key" or "write_key" than the interfaced printer and the SNMP requests fail. Now appropriate error messages are produced instead of the Traceback.

MWMatthieuBR, 2024-10-12:
I test with your new version of the project and your answer is good i don't have the good read_key :
ex : 
````DEBUG:root:EEPROM_DUMP Maintenance required level of 1st waste ink counter:
  ADDRESS: 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.54.0
  OID: 54=0x36
DEBUG:root:  TAG: OctetString
  RESPONSE: b'\x00||:41:NA;\x0c'
INFO:root:Invalid read key.
````

But i don't know how to get it. Do you know how ? thanks !

Ircama, 2024-10-13:
> But i don't know how to get it. Do you know how ? thanks !

Version 3 includes the "Detect Access Keys" button, which attempts to discover the *read_key* and *write_key* of the printer.

PUN-app, 2024-11-13:
For the ET 2850 using v5.1(IP present but hidden with `x`)...
[INFO] Detecting printers... (this might take a while)
[INFO] Found printer 'EPSON ET-2850 Series' at x.x.x.x (hostname: Unknown)
[ERROR] Printer model unknown.
[INFO] Starting the access key detection, please wait for many minutes...
[INFO] Detecting the read_key...
[ERROR] Could not detect read_key.

MWMatthieuBR, 2024-11-16:
Same as @PUN-app using the last version, can't detect the read_key..

nannys-hedgehog, 2025-01-19:
@Ircama  I'm just here to report success on an Epson ET-2825, which is the same modell as ET-2820. Thank you for all the effort you took into this project.
To get it working with the ET-2825 I had first to FW downgrade the printer to FW XE18MB from XE120B via WicResetUtility, since FW XE120B does not support EEPROM access via WIFI but FW XE18MB does. Got the printer definitions imported into epson_print_conf from [Codeberg](https://codeberg.org/atufi/reinkpy) via toml file.

Actually I did reset it at first via Linux and the https://codeberg.org/atufi/reinkpy project. Afterwards I got carried away and explored if it isn't possible via epson_print_conf too (I'm running it on a Win7 machine) and found the solution via firmware downgrade. The WicResetUtility I used was version 5.90.

mikuhl-dev, 2025-05-04:
For ET-2850 I used WIC Reset Utility to install firmware (there was only one version available), which allowed me to reset the waste ink levels after loading the TOML from https://codeberg.org/atufi/reinkpy/raw/branch/main/reinkpy/epson.toml. Printer was plugged in with USB, not sure if that made a difference.

mmbarkel-cpu, 2025-10-27:
Used v7.0 but it it said printer model unknown (L3251)... it could not reset.. PLEASE HELP

Ircama, 2026-09-22:
Version 8.0.0 introduces preliminary USB support.

yonggamer, 2026-11-03:
when are the new printers suported?

ItachiUchiha-IU, 2026-11-10:
> For ET-2850 I used WIC Reset Utility to install firmware (there was only one version available), which allowed me to reset the waste ink levels with epson_print_conf after loading the TOML from https://codeberg.org/atufi/reinkpy/raw/branch/main/reinkpy/epson.toml. Printer was plugged in with USB, not sure if that made a difference.

I confirm I just succeeded in resetting the waste ink levels for ET-2850 following your advice!
I will just specify that being plugged in with USB is necessary for the usage of WicReset, while an internet connection is necessary for WicReset to download the old firmware (`XJ17MB 17/11/2022`, which is the one that allows us to read the EEPROM data and that WicReset lists in their site as well: https://wic-reset.com/printer/epson-et-2850); `epson_print_conf` was used through WIFI.

Personally, I only got a little trouble in the beginning with getting the USB connection to work properly, but WicReset seems to be pretty safe as it keeps giving u errors and explaining everything that doesnt work and avoiding taking any action if things arent working properly (there's also a "test the printer connection" option specifically for that purpose).
Ill also add that, just to be ultra-safe and because it was advised by the assistant previously, I made a backup of the EEPROM (`Save` iirc) before downgrading the firmware.

I'll also link the "tutorial" that someone made on reddit by following your advice:
https://www.reddit.com/r/printers/comments/1r8wp13/tutorial_reset_epson_et2850_waste_ink_counter/
and also comment this same message there.