Ircama/epson_print_conf #27: ET-2800 - Detected but not in drop down

RhondaSS, 2024-10-03:
I have the ET-2800 and its detected via IP address and shows all status info but not included in drop down.

Ircama, 2024-10-03:
Printer not included in the configuration.

I have no time to provide usage support. 

You can [try to integrate it yourself](https://github.com/Ircama/epson_print_conf/?tab=readme-ov-file#example-to-integrate-new-printers) at your risk.

topdeckg, 2026-08-25:
> I ran into similar issues. I was able to roll back the firmware to pre 2023 then this tool worked beautifully. Thanks!

How did you do this? I need to do the same thing.

mgtoomer, 2026-08-25:
> > I ran into similar issues. I was able to roll back the firmware to pre 2023 then this tool worked beautifully. Thanks!
> 
> How did you do this? I need to do the same thing.

Sorry,  I don't recall for sure but I think it was all with the tool

topdeckg, 2026-08-25:
> > > I ran into similar issues. I was able to roll back the firmware to pre 2023 then this tool worked beautifully. Thanks!
> > 
> > 
> > How did you do this? I need to do the same thing.
> 
> Sorry, I don't recall for sure but I think it was all with the tool

What tool? This thread is literally the only source I can find that such a thing is even possible

mgtoomer, 2026-08-25:
> What tool? This thread is literally the only source I can find that such a thing is even possible

The tool that this page is for

https://github.com/Ircama/epson_print_conf/releases/tag/v7.1.0

topdeckg, 2026-08-25:
> > What tool? This thread is literally the only source I can find that such a thing is even possible
> 
> The tool that this page is for
> 
> https://github.com/Ircama/epson_print_conf/releases/tag/v7.1.0

You reset the firmware with this tool? How did you do that?

Ircama, 2026-09-22:
Version 8.0.0 introduces preliminary USB support.