Ircama/epson_print_conf #29: ET-4800 - Unable to clear waste levels

CarlG17, 2024-10-07:
Hi, I've been trying to clear the waste ink levels on an ET-4800, but I'm running into an error of an invalid printer model. I've added the model to the dropdown through the method listed in the readme, and the GUI can read the status of printer correctly, but when I try to clear the levels I get:
" [ERROR] Cannot read EEPROM values for "Raw waste reset": invalid printer model selected."

Any advice on how to get round this?
[image]

Ircama, 2024-10-08:
This looks the same as #26.

The ET-4800 printer is not part of epson_print_conf.py. Anyway, you are probably using a wrong printer model or a wrong printer definition, with a different "read_key" than the interfaced printer.

Run the UI with the -d command-line option (debug). You should see a number of "INFO:root:Invalid read key.".

Ircama, 2024-10-09:
> Running with the debug option does show a lot of invalid read keys

Strange.

Try using Python to run the application from the command line and check the output produced by the `-d` debug option when pressing "Reset Waste Ink levels".

Be sure to use the latest version of the application.

In the UI, select the drop-down list of the models and press F2 in correspondence to your model. The status box will show the printer definition values. Select the status box, right button of the mouse, then copy all data and paste them here.

jakusotsu, 2024-10-14:
> Strange.
>
> Try using Python to run the application from the command line and check the output produced by the `-d` debug option when pressing "Reset Waste Ink levels".
>
> Be sure to use the latest [...]

I have this same model and running into the same issue:

**Result of F2**
`
"ET-4800": {"read_key": [74, 54],"write_key": b"Maribaya","main_waste": {"oids": [48, 49, 47], "divider": 63.46},"borderless_waste": {"oids": [50, 51, 47], "divider": 34.16},"third_waste": {"oids": [252, 253, 254], "divider": 13.0},"raw_waste_reset": {48: 0, 49: 0, 47: 0, 52: 0, 53: 0, 54: 94, 50: 0, 51: 0, 55: 94, 28: 0, 252: 0, 253: 0, 254: 0, 255: 94},"stats": {"Manual cleaning counter": [90],"Timer cleaning counter": [89],"Power cleaning counter": [91],"Total print pass counter": [133, 132, 131, 130],"Total print page counter": [776, 775, 774, 773],"Total scan counter": [1843, 1842, 1841, 1840],"Total scan counter % (ADF)": [1855, 1854, 1853, 1852],"Ink replacement counter %-- Black": [554],"Ink replacement counter %-- Cyan": [555],"Ink replacement counter %-- Magenta": [556],"Ink replacement counter %-- Yellow": [557],"Maintenance required level of 1st waste ink counter": [54],"Maintenance required level of 2nd waste ink counter": [55],"Maintenance required level of 3rd waste ink counter": [255],},"serial_number": range(1604, 1614),}`

**Result of Printer Status**
`
"ET-4800": {
    "cartridges": ["\x00ia:;"],
    "firmware_version": "XD08N9 08 Sep 2023",
    "printer_status": {
        "ready": True,
        "status": (4, "Idle (ready to print)"),
        "paper_path": b"\x01\x00",
        "unknown": [("0xa", b"\x11\x00\x04"), ("0x28", b"\xff\x01\x00\x00"), ("0x4b", b"\x01")],
        "ink_level": [(1, 0, "Black", "Black", 70), (5, 3, "Yellow", "Yellow", 86), (4, 2, "Magenta", "Magenta", 86), (3, 1, "Cyan", "Cyan", 86)],
        "loading_path": "010A4E",
        "cancel_code": "No request",
        "jobname": "Not defined",
        "paper_count": "error",
        "maintenance_box_1": "not full (0)",
        "maintenance_box_reset_count_1": 0,
        "maintenance_box_2": "not full (0)",
        "maintenance_box_reset_count_2": 0,
        "serial_number_info": "X8GN248904",
    },
    "serial_number": "?",
    "snmp_info": {
        "Model": "EPSON ET-4800 Series",
        "Epson Printer Name": "ET-4800",
        "Model short": "ET-4800 Series",
        "Epson Personal Name": "ET-4800 Series",
        "EEPS2 firmware version": "EEPS2 Hard Ver.1.00 Firm Ver.0.19",
        "Epson Version number": "05.19.XD08N9.FY20",
        "Descr": "EPSON Built-in 11b/g/n & 10/100 Print Server",
        "UpTime": "00:28:45",
        "Name": "EPSONF63430",
        "MAC Address": "38-1A-52-F6-34-30",
        "Print input": "Rear Auto Sheet Feeder",
        "Lang 1": "ESCPL2",
        "Lang 2": "BDC",
        "Lang 3": "D4",
        "Lang 4": "ESCPR2",
        "Emulation 1": "ESC/P2",
        "Emulation 2": "Remote Command",
        "Emulation 3": "other",
        "Emulation 4": "ESC/P-R",
        "Total printed pages": "80",
        "IP Address": "",
        "IPP_URL": "http://192.168.87.73:631/ipp/print",
        "WiFi": "LEEFAMILY",
        "MAC Addr": "38-1A-52-F6-34-30",
        "Epson device id": "MFG:EPSON;CMD:ESCPL2,BDC,D4,D4PX,ESCPR2,END4,GENEP;MDL:ET-4800 Series;CLS:PRINTER;DES:EPSON ET-4800 Series;CID:EpsonRGB;FID:FXA,DPN,WFA,ETA,AFA,DAN,WRA;RID:20;DDS:022500;ELG:1162;SN:5838474E3234383904;",
        "Power Off Timer": "\x00ot:;\x0c",
    },
    "stats": {
        "Manual cleaning counter": None,
        "Timer cleaning counter": None,
        "Power cleaning counter": None,
        "Total print pass counter": None,
        "Total print page counter": None,
        "Total scan counter": None,
        "Total scan counter % (ADF)": None,
        "Ink replacement counter %-- Black": None,
        "Ink replacement counter %-- Cyan": None,
        "Ink replacement counter %-- Magenta": None,
        "Ink replacement counter %-- Yellow": None,
        "Maintenance required level of 1st waste ink counter": None,
        "Maintenance required level of 2nd waste ink counter": None,
        "Maintenance required level of 3rd waste ink counter": None,
    },
}
`

Ircama, 2024-10-15:
You might try the "Detect Access Keys" button included in Version 3, which attempts to discover the read_key and write_key of the printer.

pinpoint90, 2024-10-27:
> This looks the same as #26.
>
> The ET-4800 printer is not part of epson_print_conf.py. Anyway, you are probably using a wrong printer model or a wrong printer definition, with a different "read_key" [...]

Is it possible to add support for ET 4800?

pinpoint90, 2024-11-08:
Have the same model. When runnuing "Detect Access Keys" I end up wiuth this error:

[ERROR] More than one pattern appears to be a serial number:
[ERROR] - found pattern "X8GT010188" at address 793
[ERROR] - found pattern "X8GT010188" at address 1604
[ERROR] Could not detect serial number.

Florgo88, 2024-11-26:
Hi i'm having the same problem. Can someone fix this please? thanks

Ircama, 2024-11-26:
> Hi i'm having the same problem. Can someone fix this please? thanks

Did you try the "Detect Access Keys" operation? Can you provide its result? Can you also provide the output of F2 and F3?

dethlejd-ttc, 2024-12-30:
Here are the values for my ET-4800 in ui.py

F2:
"ET-4800": {
    "read_key": [255, 254],
    "write_key": b"Maribaya",
    "main_waste": {"oids": [48, 49, 47], "divider": 63.46},
    "borderless_waste": {"oids": [50, 51, 47], "divider": 34.16},
    "third_waste": {"oids": [252, 253, 254], "divider": 13.0},
    "raw_waste_reset": {48: 0, 49: 0, 47: 0, 52: 0, 53: 0, 54: 94, 50: 0, 51: 0, 55: 94, 28: 0, 252: 0, 253: 0, 254: 0, 255: 94},
    "stats": {
        "Manual cleaning counter": [90],
        "Timer cleaning counter": [89],
        "Power cleaning counter": [91],
        "Total print pass counter": [133, 132, 131, 130],
        "Total print page counter": [776, 775, 774, 773],
        "Total scan counter": [1843, 1842, 1841, 1840],
        "Total scan counter % (ADF)": [1855, 1854, 1853, 1852],
        "Ink replacement counter %-- Black": [554],
        "Ink replacement counter %-- Cyan": [555],
        "Ink replacement counter %-- Magenta": [556],
        "Ink replacement counter %-- Yellow": [557],
        "Maintenance required level of 1st waste ink counter": [54],
        "Maintenance required level of 2nd waste ink counter": [55],
        "Maintenance required level of 3rd waste ink counter": [255],
    },
    "serial_number": [range(793, 803), range(1604, 1614)],
    "wifi_mac_address": range(1920, 1926),
}

F3:
"ET-4800": {
    "Printer model": "ET-4800",
    "Read sequence": "255.254",
    "Hex read sequence": "ff fe",
    "Value of the 'write_key'": b"Maribaya",
    "Write string": "Nbsjcbzb",
    "Write sequence": "78.98.115.106.99.98.122.98",
    "Hex write sequence": "4E 62 73 6A 63 62 7A 62",
    "OID - Read address 0": "1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.255.254.65.190.160.0.0",
    "OID - Write value 0 to address 0": "1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.16.0.255.254.66.189.33.0.0.0.78.98.115.106.99.98.122.98",
}

Ircama, 2024-12-30:
> Here are the values for my ET-4800 in ui.py
> ...

Thanks. The [ET-4800](https://github.com/Ircama/epson_print_conf/blob/89ffe449996d6bebb83cfa09f242b83054770a34/epson_print_conf.py#L369-L397) model does not need to load or import external configuration archives, as the configuration is already embedded in the program. The read key should be [74, 54](https://github.com/Ircama/epson_print_conf/blob/89ffe449996d6bebb83cfa09f242b83054770a34/epson_print_conf.py#L370), while you report 255, 254. I would like to understand why these values changed.

Did you run the latest version of the program v5.3.6?

Did you run it from Python? Or the *epson_print_conf.exe* executable?

Can you describe the way you used it? Did you press "Detect configuration" or "Detect Access Keys" before testing F2 and F3, or just F2 and F3 after running the program?

dethlejd-ttc, 2024-12-30:
Environment:
Windows
Python v3.10.7
git windows

Cloned most recent from git and pulled up to date

Running ui.py with the -d flag via gitbash commandline. (e.g. python3 ui.py -d) and can see debug output in the shell.

Find printers finds the ET-4800 well enough (on it's assigned IP address)

I select the correct dropdown for the ET-4800 (and that IS the model of printer I have) and apply the IP address.

If I mash the "Detect Access Keys" it starts access key detection, runs for a while and then errors with: Could not detect read_key. There is no output to the invoking console.

At this point, F2 and F3 do not switch UI context and show their data.

If I press "Printer Status" invoking console throws a ton of "Invalid Read Key" errors, but F2 and F3 start returning UI data.

Ircama, 2024-12-30:
> I select the correct dropdown for the ET-4800 (and that IS the model of printer I have) and apply the IP address.

Can you confirm that, if you do not press "Detect Access Keys", then F2 and F3 show the default read key [74, 54], but you fail to reset the ink waste with this model?

dethlejd-ttc, 2024-12-30:
If I restart ui.py, and manually enter the Printer Model and IP address (skipping the "Detect Printers"), and press F2, the read_key values are 74,54.

Oddly, the read sequence returned from F3 is: 74.54 (note the '.' and not a ','), if that is valuable information.

If I then select the Reset Waste Ink Levels, the invoking console throws a mess of:

`$ python ui.py -d
DEBUG:root:EEPROM_DUMP Raw waste reset:
  ADDRESS: 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.48.0
  OID: 48=0x30
DEBUG:root:  TAG: OctetString
  RESPONSE: b'\x00||:41:NA;\x0c'
INFO:root:Invalid read key.
`
and the UI itself throws:

`[ERROR] Cannot read EEPROM values for "Raw waste reset": invalid printer model selected: ET-4800.
`

Ircama, 2024-12-31:
`RESPONSE: b'\x00||:41:NA;\x0c'` means that there is a problem in reading the EEPROM with that OID.

I assume that also `snmpget -v1 -d -c public <IP_address> 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.48.0` fails.

I would add a test: at the moment, before writing data to the EEPROM, I attempt to read data from it in order to verify the EEPROM access and log useful information to be able to reverse the operation; for some models, this read operation fails (unknown reason). Anyway, it is possible to write directly to the EEPROM without the preliminary reading operation. For instance, to reset the waste ink levels, do the following:

- open the app
- load the appropriate printer configuration file (from the menu)
- select the correct printer model
- select the correct IP address
- from the menu, Settings > Show printer parameters of the selected model (F2); press "Printer status" to verify a valid output from the printer
- from the menu, Settings > Debug Shell (F7)
- at the prompt, enter the following (single line):

```python
for oid, value in self.printer.parm["raw_waste_reset"].items(): self.printer.snmp_mib(self.printer.eeprom_oid_write_address(oid, value, label="raw_waste_reset"), label="raw_waste_reset")
```

paste the output here

dethlejd-ttc, 2024-12-31:
First, snmpget from a nearby linux host (not from the windows machine)

snmpget -v1 -d -c public 192.168.5.115 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.48.0
No log handling enabled - using stderr logging

Sending 63 bytes to UDP: [192.168.5.115]:161->[0.0.0.0]:0
0000: 30 3D 02 01  00 04 06 70  75 62 6C 69  63 A0 30 02    0=.....public.0.
0016: 04 79 C3 1A  CF 02 01 00  02 01 00 30  22 30 20 06    .y.........0"0 .
0032: 1C 2B 06 01  04 01 89 60  01 02 02 2C  01 01 02 01    .+.....`...,....
0048: 7C 7C 07 00  4A 36 41 81  3E 81 20 30  00 05 00       ||..J6A.>. 0...

Received 74 byte packet from UDP: [192.168.5.115]:161->[0.0.0.0]:0
0000: 30 48 02 01  00 04 06 70  75 62 6C 69  63 A2 3B 02    0H.....public.;.
0016: 04 79 C3 1A  CF 02 01 00  02 01 00 30  2D 30 2B 06    .y.........0-0+.
0032: 1C 2B 06 01  04 01 89 60  01 02 02 2C  01 01 02 01    .+.....`...,....
0048: 7C 7C 07 00  4A 36 41 81  3E 81 20 30  00 04 0B 00    ||..J6A.>. 0....
0064: 7C 7C 3A 34  31 3A 4E 41  3B 0C                       ||:41:NA;.

SNMPv2-SMI::enterprises.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.48.0 = Hex-STRING: 00 7C 7C 3A 34 31 3A 4E 41 3B 0C

Seems to return somewhat useful data.

dethlejd-ttc, 2024-12-31:
Second step: I'm not sure what you mean by "load the appropriate printer configuration file (from the menu)" as that doesn't seem to be a menu choice; do you want me to export the config as a PICKLE file and reimport it?

dethlejd-ttc, 2024-12-31:
If, however, I skip the load step, manually select the ET-4800 from the Printer Model drop down, apply the correct IP address in the IP Address field, and then run the debug console with the supplied for statement, I get:

('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')
('OctetString', b'\x00||:42:NA;\x0c')

Ircama, 2024-12-31:
> 0064: 7C 7C 3A 34 31 3A 4E 41 3B 0C ||:41:NA;.
> ...
> Seems to return somewhat useful data.

`||:41:NA;` means that the OID cannot be used to read that EEPROM.

_____

> Second step: I'm not sure what you mean by "load the appropriate printer configuration file (from the menu)" as that doesn't seem to be a menu choice; do you want me to export the config as a PICKLE file and reimport it?

Ignore this, sorry.

_____

> ('OctetString', b'\x00||:42:NA;\x0c')
> ...

Thanks for the test; `||:42:NA;` means that unfortunately the ET-4800 with your firmware cannot be managed by the current algorithm.

Ircama, 2024-12-31:
The code that worked with an ET-4800 in issue https://github.com/Ircama/epson_print_conf/issues/39 was possibly interfacing an older firmware.

dethlejd-ttc, 2024-12-31:
I did apply a firmware update within the last couple of weeks, perhaps Epson tires of people trying to fix things themselves.

dgerber, 2024-12-31:
> I did apply a firmware update within the last couple of weeks, perhaps Epson tires of people trying to fix things themselves.
As for other models, you may still be able to fix it via USB.

Florgo88, 2024-12-31:
> I did apply a firmware update within the last couple of weeks, perhaps Epson tires of people trying to fix things themselves.
> As for other models, you may still be able to fix it via USB.

Hi how can we fix it via USB?
I also have F3 255.254

dgerber, 2024-12-31:
> Hi how can we fix it via USB?
> I also have F3 255.254
Both the original reink and [reinkpy](https://codeberg.org/atufi/reinkpy) support USB.

Florgo88, 2024-12-31:
> Hi how can we fix it via USB? I also have F3 255.254
> Both the original reink and [reinkpy](https://codeberg.org/atufi/reinkpy) support USB.

Thank you very much. i will try that

Ircama, 2024-12-31:
Thanks @dgerber for letting me know that USB still works. I have never tested it (and I might not even have the cable). Does USB use the same SNMP [byte sequences](https://github.com/Ircama/epson_print_conf?tab=readme-ov-file#byte-sequences)?

dgerber, 2024-12-31:
> Thanks @dgerber for letting me know that USB still works. I have never tested it (and I might not even have the cable). Does USB use the same SNMP [byte sequences](https://github.com/Ircama/epson_print_conf?tab=readme-ov-file#byte-sequences)?

I've not tested on ET, but it did work on an XP-Series, as we saw on
another ticket here.

It seems byte sequences are the same, just the encoding is different
over USB or SNMP.

Ircama, 2024-12-31:
> I did apply a firmware update within the last couple of weeks, perhaps Epson tires of people trying to fix things themselves.

New firmware versions might be likely being developed to address a security issue arising from such use of the SNMP access protocols.

The exploitation of security keys to access the printer's EEPROM, along with the inevitable dissemination of the access keys, including the APIs that facilitate their use, exposes printers to the risk of unauthenticated network access by malicious actors. This could result in irreversible changes to the EEPROM, compromising the device's functionality.

Ultimately, the community’s contribution to developing open-source tools and open data for accessing the printer EEPROM has shed light on a longstanding security flaw in these printers — one that existed but was previously less recognized.

dethlejd-ttc, 2025-01-01:
I can confirm, however that after pulling the reinkpy repo and connecting directly to USB from a linux host, that the ET-4800 will allow its cleaning/inktank counters to be reset.

Florgo88, 2025-02-07:
> I can confirm, however, that after pulling the reinkpy repo and connecting directly to USB from a linux host, that the ET-4800 will allow its cleaning/inktank counters to be reset.

Hi, can you explain how i can make this work? thanks a lot

dethlejd-ttc, 2025-02-11:
> > I can confirm, however, that after pulling the reinkpy repo and connecting directly to USB from a linux host, that the ET-4800 will allow its cleaning/inktank counters to be reset.
> 
> Hi, can you explain how i can make this work? thanks a lot

So, if I recall, here's what I did:

First, I got a replacement set of printer tampons (you can find generic replacements on the internet instead of buying the Epson kit).

Then, I installed raspbian (Pi OS installer, grabbed the current ISO, installed on the microSD card via a USB bridge written to by Rufus) on a Raspberry Pi to get an OS that could easily run the code from the reinkpy git repo, connected it to my Ethernet network (so I could shell into it) and then powered it up.

ssh'd to the Pi, installing git and made sure that I had the correct cable (USB-A > USB-B), and that when connected via the USB cable, the Pi running debian recognized the USB device.

I pulled the reinkpy repo from git, (git clone local) and executed the reinkpy code (it has a console UI). Pointed the tool at the /dev/usb device and let it do it's thing. It successfully navigates the firmware and zero'd out the inkpad timer.

I got a couple sets of the replacement pads, so I can do it again in 5,000-some-odd pages of printing.

The replacement process itself was a little finicky (and messy; use some disposable latex gloves and try and not tip the reservoir all over the place) and the the access to its little cubbyhole involved some tiny, easily losable screws. It also felt as if I was almost using too much force on the plastic door and tank. You might consider getting one of the replacement kits that comes with an extra tank, as well, just in case.

Other than that? Easy-peasy.

Remember to run the printer through it's housekeeping routines (head cleaning, alignment, loading the ink to the print heads, etc.)

Ircama, 2026-09-09:
Your printer model is available in the Epson printer configuration database and can be imported into this program as described in the README: https://github.com/Ircama/epson_print_conf#how-to-import-an-external-printer-configuration-db

closed 2026-09-09 by Ircama

Ircama, 2026-09-22:
Version 8.0.0 introduces preliminary USB support.

Florgo88, 2026-09-22:
Hi, 

I tried the SNMP reset but it doens't work. Should I try the USB method?

Ircama, 2026-09-22:
Yes, the USB method should work.