Ircama/epson_print_conf #32: Adding New Printers from epson.toml

aramis201, 2024-10-20:
Hello!

Tyvm for this program was able to reset an ET-2720 it's working great. 

Now I'm trying to do an ET-2850 but didn't find anything in the attachment file though there's this other program i was trying that has this epson.toml file specifically this part:

[[EPSON]]
rkey = 0x364A
wkey = "Bsboujgp"
wkey1 = "Arantifo"
models = ["ET-2850", "ET-2851", "ET-2853", "ET-2855", "L4260", "L4261", "L4263", "L4265", "L4266", "L4267", "L4268", "L4269", "Workforce ST-C2100"]
mem = [
  { addr = [0x1C,0x34,0x35,0x36,0x37,0xFF] , desc = "Waste counters (?)" , reset = [0x00,0x00,0x00,0x5E,0x5E,0x5E] },
  { addr = [0x2F] , desc = "Waste counter" },
  { addr = [0x30,0x31] , desc = "Waste counter" },
  { addr = [0x32,0x33] , desc = "Waste counter" },
  { addr = [0xFC,0xFD] , desc = "Waste counter" },
  { addr = [0xFE] , desc = "Waste counter" },
]
rlen = 2
wlen = 2
mem_high = 0x7FF

anyway I can use this to create the pickle file to reset this printer?

TYVM

Ircama, 2024-10-20:
This repository is focused on Python code development, I'm sorry but I have no time to provide usage support and add/test new printers.

The [README](https://github.com/Ircama/epson_print_conf/?tab=readme-ov-file#key-features) includes all explanations on doing the work yourself. You might check [the related chapter](https://github.com/Ircama/epson_print_conf/?tab=readme-ov-file#example-to-integrate-new-printers).

Ircama, 2024-10-21:
You can use this command to process you fille:

```
python3 parse_devices.py -T -c epson.toml -i -l 200
```

aramis201, 2024-10-22:
> You can use this command to process you fille:
> 
> ```
> python3 parse_devices.py -T -c epson.toml -i -l 200
> ```

Thank you! I'll try that tonight let you know how it goes.

closed 2025-01-16 by github-actions[bot]