Ircama/epson_print_conf #35: L3250 resetting waste ink issue

klifinpol, 2024-10-31:
Hello, I've tried using your tool to reset the ink levels on one of my L3250 and I have ran into multiple errors. 

When I try to ''Reset Waste Ink Levels'' it pops up with an error saying ''cannot readd EEPROM values for ''Raw waste rest'': invalid printer model selected. So I saw that somone had a simmilar issue and it was suggested to use the ''Detect Access Keys'' which I tried and it crashed the program, similarly when I tired using Detect configuration it also spit out an error saying ''int() can't convert non-string with explicit base'' 

Due to the crashes I attempted to use the the comand-line tool, however similarly after the process was completed it poped up with and error saying ''Cannot found Read_key''

I'm not a coder so apologise if I do not know what these error codes mean exactly however I hopefully have provided enough data to be able to solve this issue... I hope.

[image]
[image]

Ircama, 2024-10-31:
Thanks this helped correct something.

Please try running the command line tool with the `-d` option and upload the output.

Example:

```
python3 epson_print_conf.py --detect-key -a 192.168.178.29 -m L3250 -d >trace.log 2>&1
```

You should get a file named *trace.log* that you can browse and upload.

klifinpol, 2024-10-31:
here is the trace log, hope it helps
[trace.log](https://github.com/user-attachments/files/17592870/trace.log)

klifinpol, 2024-10-31:
Sorry about my previous comment I used the incorrect adress (mine ends with .32) here are the correct .log files.
I took the liberty of trying two other commands aswell. One was the --reset_waste_ink and -i (i was unable to locate the ''detect configuration'' equivalent in the command line). I truely hope this helps you find and solve the issue as my printer is currently an expensive paperweight...

(I had to add the bruteforce attempt in two parts as the log file on its own was too large)

[trace_BruteForce_Part1.txt](https://github.com/user-attachments/files/17594077/trace_BruteForce_Part1.txt)
[trace_Bruteforce_Part2.log](https://github.com/user-attachments/files/17594076/trace_Bruteforce_Part2.log)
[trace_info.log](https://github.com/user-attachments/files/17594037/trace_info.log)
[trace_wasteRest.log](https://github.com/user-attachments/files/17594038/trace_wasteRest.log)

Ircama, 2024-11-01:
It fails to read values from the EEPROM. From your logs, the following is strange:

```
  ADDRESS: 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.0.0
  OID: 0=0x0
  TAG: OctetString
  RESPONSE: b'\x00||:41:NA;\x0c'
```

Same issue as https://github.com/Ircama/epson_print_conf/issues/26#issuecomment-2408652905.

If you have a Linux system (or Windows Subsystem for Linux), you might run the following commands and paste the outputs:

```
snmpget -v1 -d -c public 192.168.178.32 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.0.0
snmpget -v1 -d -c public 192.168.178.32 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.48.0
snmpget -v1 -d -c public 192.168.178.32 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.48.1
snmpget -v1 -d -c public 192.168.178.32 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.255.0
```

There are also [binaries for Windows](https://netcologne.dl.sourceforge.net/project/net-snmp/net-snmp%20binaries/5.7-binaries/net-snmp-5.7.0-1.x86.exe?viasf=1) which include *snmpget.exe*, that runs with the same arguments.

The answer should include these two strings `@BDC PS` and `EE:`

Ircama, 2024-11-01:
Please also try these:

```
snmpget -v1 -d -c public 192.168.178.32 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.1.0
snmpget -v1 -d -c public 192.168.178.32 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.74.54.65.190.160.0.1
```

klifinpol, 2024-11-01:
[image]

Here you go. If you'd like I'd be willing to hop on discord or somthing similar to try and help resolve this issue.

Ircama, 2024-11-01:
Thanks. Unfortunately, there is no way to manage this printer model at the moment; the only thing which can be done is temporarily removing this model from the configuration. I will ask you for more tests as soon as I have other ideas.

klifinpol, 2024-11-01:
I called a few of my programmer friends to help fix my printer... We are trying a few ideas; aporximatley 65k ideas
[image]

klifinpol, 2024-11-01:
... we ran out of ideas... (our attempt didn't work)

Ircama, 2025-07-13:
The latest version of epson_print_conf has a "Temporary reset of the ink waste counter" which I think it should work with this printer. It has to be repeated each time the printer is switched on.

fullofpride012-ops, 2025-09-08:
Hello, thank you for making this resetter. i have installed it as of 8/9/25, but it's unfortunate that the epson l3250 is not supported. i would like to be updated once the printer is good to use for this software though! thank you once again and all the best!

VXsz, 2025-10-18:
> I've been having the same issues. I found the following software. Give it a try https://www.waredata.com/epson-l3210-resetter-printer/ It worked for me.

This was shady as fuck, but it did work and was the only free working option that I kept finding. So, I gave in

I inspected all connections while running it (was mainly concerned about my local network) and beside Microsoft OS spam I saw nothing and no communication to random IPs/domains. Though I'd advice others to take as much caution and at least check with proxy software like I did as possible. I had a throwaway windows install so it wasn't as critical in terms of files and other stuff. I noticed no weird behavior, though I wonder why they protected the exe with engimeprotect, maybe hidden malware maybe something else.

BTW maintainers, I remember my printer updating like a few months ago, yet that shady tool worked even though its from 2021. So I wonder how this issue suddenly popped up here if old stuff work 🤔.

ghuser-304951, 2025-11-13:
@adasThePro Thanks bro! 🥹 ❤️

onur-kesim, 2026-08-28:
Hi — I've been working on the L3250 generation from the USB side, and I'd rather ask you one question before putting effort into a PR you may not want.

**What I have.** A pure-Python (stdlib `ctypes` only) IEEE 1284.4 / D4 transport that talks to the Windows `USBPRINT` device interface directly — no driver replacement, no libusb, no build step. It brings up D4 including the credit handshake, and reads and writes EEPROM cells. I've used it on an L3251: full 256-cell bank-0 backup, waste-counter decode, and a counter reset with read-back verification. Code: https://github.com/onur-kesim/epson-l3251-usb-reset

**Why in this thread.** You wrote here that "epson_print_conf runs correctly based on the prerequisite of EEPROM accessibility via SNMP", and that where the firmware has that disabled, the way around it would be a USB driver for updating the firmware. My unit matches the first half: the SNMP path is closed on it. The second half turned out differently — I did not need to touch the firmware. The D4 channel over `USBPRINT` still answers EEPROM reads and writes on this generation, on the secondary USB interface (MI_01 on my unit). Someone in this thread also asked directly whether it can be reset over USB; on an L3251, it can.

**My actual question is licensing, and it's yours to answer.** The EEPROM command parameters and waste-counter addresses I use come from prior reverse-engineering work, reinkpy (AGPL-3.0) among it. This project is EUPL-1.2. My own code is mine to license however you need it; I don't think the reverse-engineered constants are mine to speak for. How do you handle that here?

And separately — is a Windows USB transport something you would want in this project at all? "No" is a perfectly good answer and I'll keep it as a standalone tool. Either way I have the printer in front of me, so if test data would be more useful to you than code, tell me what to measure and I'll run it.

endafk, 2026-08-29:
Hi all, i finally got my L3250 permanently reset last night, so let me write down everything that worked in case it helps the people still stuck here. @vista101-lab asked back in May if this can be reset over USB and nobody answered, the answer is yes it can. Also i see some of you ended up running that waredata resetter on a throwaway Windows install, you really dont need to do that.

**The short version: the EEPROM on this printer is not locked. Only the SNMP path to it is locked.** Over USB it opens right up. That is why this tool says L3250 is incompatible, it talks SNMP, and it is correct about SNMP but it made us all think the printer itself was sealed. It isnt.

### how i confirmed the SNMP side is really dead

I didnt want to guess so i brute forced it. Every single one of the 65536 possible two byte read keys, sent to the `||` EEPROM command over SNMP. All 65536 came back `||:41:NA;`. Took about 15 minutes. So dont waste your time on "Detect Access Keys" on this model, it will not find anything, there is nothing to find on that channel. My firmware is 05.22.XF26P8.

(side note, that also explains @silvesterkeli1-max crash. the key detection has a bug where it only tests 65280 of the 65536 keys, it uses `itertools.permutations` which never gives you a pair of the same byte, so keys like [7,7] can never be found. i sent a fix in #129)

### what actually worked, step by step

The tool that does it is **reinkpy** by Daniel Gerber, https://codeberg.org/atufi/reinkpy . He already had our printer figured out, the key and the memory addresses have been sitting in his `epson.toml` this whole time under the L3250 group. All the credit here is his, i just got it running on Windows.

1. Plug the printer in with a **USB cable**. Network will not work for this, it has to be USB.

2. Get reinkpy, it is not on pypi:
```
git clone https://codeberg.org/atufi/reinkpy.git
cd reinkpy
pip install -e .
pip install pyusb libusb
```

3. Two Windows gotchas that cost me hours:

   **a)** pyusb will happily load the wrong libusb DLL. Mine picked the **arm64** one and just kept saying `NoBackendError: No backend available` forever. You have to point it at the x86_64 one explicitly, something like:
```python
import usb.backend.libusb1, libusb, os
dll = os.path.join(os.path.dirname(libusb.__file__),
                   "_platform", "_windows", "x86_64", "libusb-1.0.dll")
be = usb.backend.libusb1.get_backend(find_library=lambda x: dll)
```
   **b)** USB interfaces 0 and 1 are held by the Windows Epson print driver, if you try to claim them you get `NotImplementedError: Operation not supported or unimplemented on this platform`. **Interface 2 is free** (class 255, subclass 170) and that is the one that works. So just loop over the interfaces and use the one that lets you claim it.

   Good news is you do **not** need Zadig, you do **not** need to replace any driver, and your normal Epson driver keeps working the whole time. VID/PID is `04B8:118A`.

4. Back up your EEPROM before you touch anything. Dump all 2048 addresses to a json file and keep it somewhere safe. If something goes wrong you want to be able to put it back. I cannot stress this enough.

5. Then reinkpy's own `reset_waste()` does the rest. It writes these 14 cells:

```
0x1C 0x2F 0x30 0x31 0x32 0x33 0x34 0x35 0xFC 0xFD 0xFE  ->  0x00
0x36 0x37 0xFF                                          ->  0x5E
```

Do it atomically and read back after to check it took.

### how you know it worked

Before the reset my main counter at `0x30`/`0x31` read `0x18CA`, thats 6346. The divider for this family is 63.46, and 6346 / 63.46 is exactly 100.00%. Thats your printer telling you the pad is full. After the write it read 0. Power cycled it, checked again, still 0. Thats the part that matters, the temporary reset does not survive a power cycle but this one does because its actually in the EEPROM.

Also `maintenance_box_reset_count_1` went from 2 to 0. That one is a seperate counter that just tallies how many times someone has overridden, it is not what bricks you.

### about the "Temporary Reset Waste Ink Levels" button

The README has that line struck through saying it should still work on incompatible models. It doesnt, and i found out why. It gets the serial number by reading it **out of the EEPROM**, which on our printers is the exact thing thats locked, so it ends up hashing `None` and the printer answers `rw:01:NA;`. But the `rw` command only needs the serial, it doesnt need a read key at all, and the status block gives you the serial in plain text anyway. So its a two line fix. Mine went from `NA` to `rw:01:OK;` straight away. Thats also in #129 if @Ircama wants it.

So even if you cant be bothered with the USB stuff, that button can be made to work for a temporary reset to get you printing today.

### please actually clean the pad

Serious bit. That counter is not made up, it is tracking a real absorbent pad soaking up real ink. I opened mine and cleaned it before i reset the counter, so when it reads 0 now its telling the truth. If you just zero the counter and keep printing you are going to get ink on your desk and eventually into the mechanism. Reset the counter after you deal with the pad, not instead of.

### @onur-kesim

Just saw your comment from yesterday, you are doing the same thing i did but properly, pure ctypes over USBPRINT with no libusb is much nicer than what i had to fight through. If your transport lands then all the Windows pain in step 3 above just disappears. Happy to test on an L3250 if that is useful to you, mine is working now so i can compare against a known good.

Hope this unsticks some people. @rbelem @vista101-lab @silvesterkeli1-max this should be everything you need.

onur-kesim, 2026-09-04:
@endafk — thanks, that was worth more than a code review. Your little-endian reading was right, and it corrected a wrong claim in my README: the divisor wasn't wrong, my decode was. What settled it on my unit was the printer's own error state — `0xCC 0x18` reads as 6348 (100.0%) little-endian and the printer *was* in the error state; `0x3B 0x18` reads as 6203 (97.7%) and it was not. Big-endian calls both of them full and cannot explain the error clearing. Fixed and pushed.

And yes to the L3250 comparison — that is the useful part, because everything below comes from a single unit. Tell me whichever output format suits you and I will match it.

**A measurement, and I think it explains the reset failures reported for this model group.** Starting from every counter at zero I printed ten bordered photo pages in one uninterrupted run — no power cycle, no manual cleaning — and diffed full bank-0 backups taken before and after:

```
0x30 0x31   0 -> 17    main counter
0x34 0x35   0 -> 17    mirror A   (in the reinkpy reset set)
0xC0 0xC1   0 -> 17    mirror B   (NOT in the reinkpy reset set)
```

Three cells, in lockstep, +17 each. That is why a reset on this generation can look like it worked and then come back: zeroing `0x30 0x31` alone leaves the value sitting in a mirror, and the firmware re-derives it at the next power-on. `0xC0 0xC1` appears to be derived rather than authoritative — my tool never writes it, yet the firmware synced it *down* to zero on the first power-on after a full-set reset. After zeroing the full spec set, all three stayed at 0 across two power cycles.

Two more numbers from the same run: **1.7 units per photo page** (17 for ten, cleaning during the run included), which against a 6346 threshold is roughly 3700 photo pages; and bordered printing left `0xFC 0xFD` at 0 throughout, so the platen pad really is fed by borderless printing.

@Ircama — this is the kind of feedback you asked for earlier in this thread, and it is hardware data rather than an opinion about tooling. I still have the printer in front of me, so if raw before/after dumps would be more useful than my summary of them, say the word. The licensing question from my earlier comment also still stands whenever you have time for it.

onur-kesim, 2026-09-14:
@endafk — I said I'd match whatever format suits you and then never actually put one up, so here it is. Ignore the shape if yours is easier; the content is the part I need.

One record per event, full bank 0 so nothing is pre-filtered:

```json
{
  "unit": "L3250",
  "firmware": "<whatever yours reports>",
  "event": "ten-bordered-photo-pages",
  "power_cycle_index": 0,
  "cells": {
    "0x30": {"before": 0, "after": 17},
    "0x31": {"before": 0, "after": 0},
    "0x34": {"before": 0, "after": 17},
    "0x35": {"before": 0, "after": 0},
    "0xC0": {"before": 0, "after": 17},
    "0xC1": {"before": 0, "after": 0}
  }
}
```

The cell values in there are mine, measured on the L3251, so they are something to disagree with rather than a template to fill in.

**The one question I need a second unit for:** does `0xC0 0xC1` move in lockstep with `0x30 0x31` and `0x34 0x35` on your L3250? On mine all three went 0 → 17 over ten bordered photo pages in a single uninterrupted run. If yours does the same, the mirror set is a property of the model group rather than of my printer — and that is the part of this thread that is still one unit wide.

A full before/after bank-0 pair is more useful than a summary if that's cheap for you, but a summary of those six cells answers the question on its own.

Use whatever you already dump. I am not asking you to run my tool: it is written and named for the L3251 and **I have never run it against an L3250**, so it should not be the path of least resistance on your hardware.

The one field I would ask you to keep is `power_cycle_index`, incremented on every off/on. That is where the reset-that-comes-back lives, and it is invisible in a single snapshot.

@Ircama — nothing needed from you on the above. The licensing question from my August comment is still open whenever you have time for it.

Ircama, 2026-09-14:
I don't have an L3250 at hand.

endafk, 2026-09-14:
I will check it tomorrow morning @Ircama

Ircama, 2026-09-22:
Version 8.0.0 introduces preliminary USB support.

Ircama, 2026-09-22:
@onur-kesim The implementation in the `epson_usb` directory temporarily reuses and extends your code.

Ideally, I think `epson-l3251-usb-reset` could eventually be split into a reusable Python library and a separate client/application. The `epson_usb` directory is a preliminary implementation of this separation, currently integrated into `epson_print_conf` for testing.

I am not intending to embed the library permanently in epson_print_conf. I would prefer to evaluate the possibility of moving it back into your project and having it developed and published via pypi.

onur-kesim, 2026-09-23:
Thanks — I had a look at `epson_usb` in v8.0.0.

One thing to flag: three docstrings point at files that are not in the package. `d4.py` cites `testler.test_eski_yeni_esdeger`, `eeprom.py` cites `testler/mutant_kos.py`, and `epson_ctrl.py` cites `testler/test_uyumluluk.py`. None of the three are in the tree, so a reader is told that a mutant run and an equivalence test back the port, and cannot run either. `mutant_kos.py` is in the fork under `testler/`. I am happy to port the three into `epson_usb/tests/` so the claims in those docstrings can actually be executed.

On the library: yes, I would take it and publish it on PyPI. What I can stand behind is what I measured on hardware — an L3251 over USB on Windows. I cannot reproduce other models, so for those I can take patches but cannot promise fixes, and I would rather not put a date on it.

One question before that: the AGPL-3.0 was chosen for an application. For a library that other projects would take as a dependency it is a heavy constraint. Would you consider a more permissive licence for the library half, with the client staying under AGPL? I can relicense my own commits, but the extraction — the backends, the bridge, the mock printer, the test suite — is yours, so that is your call as much as mine.

Ircama, 2026-09-24:
@onur-kesim thanks for pointing this out.

I believe I have fixed the three docstrings that were still pointing to files not included in the package. It was a mechanical edit, though, so I’ll double-check that everything is correct.

You can reuse and relicense the code I have modified or added in the extraction.

endafk, 2026-09-24:
Following up on my Sep 14 comment: got an L3250 (fw XF26P8, 04B8:118A) today, at 100% (`0x30/0x31` = 6346, status error `0x10`). Reset over USB on Linux with v8.0.0 `epson_usb`.
- Before: `0x30/31`=6346, `0x34/35`=9698, `0xC0/C1`=2916, `0xFC/FD`=550, `0x36`/`0xFF`=104.
- Wrote the 14-cell ET-2810 `raw_waste_reset` set (not `0xC0/C1`). After a power cycle everything stayed at 0, **and `0xC0/C1` went to 0 by itself**, same as onur's L3251. Full before-dump is available if wanted.

Ircama, 2026-09-24:
Thanks @endafk, it looks like the reset works as expected.

onur-kesim, 2026-09-24:
I ran it on 1ef1555 (Linux, Python 3.10.12, requirements.txt in a fresh venv):

- The three docstrings now resolve: tests.test_fidelity, tests/mutant_run.py and tests/test_epson_usb.py are all in the tree.
- Full suite: 66 tests, OK.
- mutant_run.py: baseline green; waste-decode caught (1 failure), write-frame caught (7 failures); both files restored, exit 0.

One thing I noticed along the way. Without the requirements installed, the same command prints "Ran 30 tests ... OK (skipped=7)". The 7 counts classes, not tests: the SkipTest is raised in setUpClass, so unittest records one skip per class and never counts the tests inside. 36 of the 66 tests did not run, and the summary looks almost clean. The mutant proof still holds there (both mutants are caught by tests that do run), so this is about what the summary tells a reader, not about the port.

It matters more after the split: in a standalone repository epson_print_conf would normally not be installed, so those seven classes would be skipped on every CI run. In the standalone package I would make CI fail when they are skipped (for example behind an environment variable), so that a green run always means all 66 ran.

onur-kesim, 2026-09-24:
Following up: the library is now on PyPI as epson-usb 0.1.0 (MIT), published from onur-kesim/epson-l3251-usb-reset: https://pypi.org/project/epson-usb/0.1.0/

- epson_usb/ was imported from 1ef1555 unchanged (checked file by file), then relicensed under MIT with your permission; the licence file credits your changes and links your comment above.
- CI installs epson_print_conf at 1ef1555 and runs the full suite: 66 tests, 0 skipped, plus mutant_run.py. If any of the seven host-program classes is skipped, CI fails.
- The hardware table in the package README lists only the L3251 (mine) and the L3250 (endafk's report above); everything else is marked as not measured.

So epson_print_conf can depend on the package instead of the in-tree copy whenever that suits you.

Ircama, 2026-09-24:
@onur-kesim two notes on https://pypi.org/project/epson-usb:

* `requires-python = ">=3.10"`: the code also runs on older Python versions. I successfully installed 0.1.0 on Python 3.9.13 with `--ignore-requires-python` and used it to drive an XP-205 over USB.

* `PyUSB` in `dependencies`: the package's own documentation still describes PyUSB as optional, including the `epson_usb[pyusb]` extra, while the published metadata declares it as a mandatory dependency.

closed 2026-09-24 by Ircama

Ircama, 2026-09-25:
@onur-kesim, two other hints:

I think `epson_usb/compat.py` may no longer be needed in the current implementation:
https://github.com/onur-kesim/epson-l3251-usb-reset/blob/main/epson_usb/compat.py

If so, this would also require reviewing:

* `epson_usb/__init__.py`
* `epson_usb/status.py`
* `epson_usb/printer.py`

In addition, it looks like the current D4 implementation hardcodes the `EPSON-CTRL` socket to `0x02`.

There are useful references for implementing D4 service discovery and multiple logical channels in:

* https://github.com/lion-simba/reink
* https://fossies.org/linux/gutenprint/src/escputil/d4lib.h

In particular, it may be possible to negotiate the socket IDs for both `EPSON-CTRL` and `EPSON-DATA`, rather than hardcoding them. `EPSON-DATA` has been observed as D4 socket `0x40` (64).

This could potentially allow `epson_usb` to support printing over USB as well, by opening the `EPSON-DATA` channel and sending the generated ESC/P2 data through it.

onur-kesim, 2026-09-25:
Both fixed in 0.1.1: https://pypi.org/project/epson-usb/0.1.1/

- PyUSB is now only in the [pyusb] extra: `pip install epson-usb` no longer installs it, `pip install "epson-usb[pyusb]"` does.
- Requires-Python is now >=3.9. CI runs the full suite on Python 3.9 and 3.10 (66 tests, 0 skipped, both mutants caught), so the `python_version >= "3.10"` marker in your requirements.txt is no longer needed if you want 3.9 users to get it.

Your XP-205 run is in the README's hardware table, quoted as you wrote it.

Thanks also for the pointers on compat.py and D4 socket discovery (reink, d4lib.h); I've noted them.

Witton-431, 2026-10-01:
**L3251 (identifies as "L3250 Series"), firmware XF26P8 (26 Aug 2025): waste ink reset over USB works**

Tested on Kali, epson_print_conf v8.1.2 with the epson_usb library, printer USB ID 04b8:118a. The pad was cleaned first. Network/SNMP EEPROM access is locked, as the README says, so I used USB. Three things were blocking it:

1. **Wrong interface auto-picked.** The printer exposes interface 0 (vendor-specific 255/255/255), interface 1 (printer class) and interface 2 (vendor-specific, subclass 170). The auto-pick takes interface 0, where D4 never answers ("D4 Init failed"). Forcing `interface=1` from a Python script negotiated **D4 revision 0x10** and EEPROM reads worked. The CLI has no flag to choose the interface, so I used the library directly.
2. **Kernel/CUPS interference.** I stopped CUPS and unloaded `usblp` (blacklisted it) before using libusb.
3. **Write key.** With the ET-2810 profile, reads matched the printer's own report (cells 0x30/0x31 = CB 18 = 6347 = 100.02% main waste). Writing the 14-cell `raw_waste_reset` set with the profile's write key (`Maribaya`) returned OK, but read-back showed every changed cell unchanged. Using the other key listed for the same model group in reinkpy's epson.toml (`Nbsjcbzb`) made every cell verify.

After the write, cells 0x1C, 0x2F-0x35, 0xFC-0xFE were 00 and 0x36, 0x37, 0xFF were 5E. I power-cycled the printer and re-read them: still zeroed, 0xC0/0xC1 had dropped to 00 on their own, and the alternating ink/paper lights were gone.

One unit, one firmware, so please treat it as a field report. I took a full EEPROM backup (bank 0) before writing.

Possible improvements, if useful: a CLI option to choose the USB interface, preferring interface 1 / trying alternatives when D4 init fails, and an L3250/L3251 profile with the working write key.

Full step-by-step write-up with the scripts is attached. It was put together with help from Claude (Anthropic), using the work of Ircama (epson_print_conf), onur-kesim (epson-usb) and atufi (reinkpy).

[epson-L3250-L3251-waste-ink-reset-linux.md](https://github.com/user-attachments/files/32910799/epson-L3250-L3251-waste-ink-reset-linux.md)

onur-kesim, 2026-10-03:
@Witton-431 thanks for the field report. epson-usb 0.1.2 (https://pypi.org/project/epson-usb/0.1.2/) adds --interface N to the example CLI and, when D4 does not answer on the automatically chosen interface, tries the next one; the fallback is covered by tests but has not been run on real hardware.