Ircama/epson_print_conf #36: XP-960 waste ink reset issue

inspinet, 2024-11-03:
**Printer model**
Epson XP-960

**Software configuration**
"XP-960" + " configuration": {
    "read_key": [40, 9],
    "write_key": b"Irisgarm",
    "main_waste": {"oids": [16, 17, 6], "divider": 93.85},
    "borderless_waste": {"oids": [18, 19, 6], "divider": 29.77},
    "raw_waste_reset": {16: 0, 17: 0, 6: 0, 52: 94, 18: 0, 19: 0, 20: 0, 21: 0, 53: 94, 493: 0},
    "stats": {
        "Timer cleaning counter": [245],
        "Total print pass counter": [99, 98, 97, 96],
        "Total print CD-R counter": [255, 254],
        "Total scan counter": [453, 452, 451, 450],
        "Maintenance required level of 1st waste ink counter": [52],
        "Maintenance required level of 2nd waste ink counter": [53],
    },
    "serial_number": range(216, 226),
}

**Printer status**
Press "Printer status" kill UI window and the following error is returned in cmdline
`C:\temp\epson_print_conf-main>python ui.py -a 192.168.2.99 -m XP-960`
`CRITICAL:root:snmp_mib invalid address: Please call .create() to construct UdpTransportTarget object`

**Detect configuration**
As above, press "Printer status" kill UI window and the following error is returned in cmd
`C:\temp\epson_print_conf-main>python ui.py -a 192.168.2.99 -m XP-960`
`CRITICAL:root:snmp_mib invalid address: Please call .create() to construct UdpTransportTarget object`

**Describe the issue**
Trying to reset ink pad counter on my Epson XP-960. Doesn't look like python code actually communicate with printer because it throw code errors first.
Tried `python epson_print_conf.py --detect-key -a 192.168.2.99 -m XP-960 -d >trace.log 2>&1`
which resulted with:
`Trying [0, 1]...`
`EEPROM_DUMP brute_force_read_key:`
`  ADDRESS: 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.0.1.65.190.160.0.0`
`  OID: 0=0x0`
`snmp_mib invalid address: Please call .create() to construct UdpTransportTarget object`

I tried to run `ui.py` from Visual Studio Code, and its starts fine so take it there is no missing modules etc. When select printer model from the list, populate correct IP address and press GET button, GUI window stop responding:
[image]
and in `epson_print_conf.py` line 1152 `UdpTransportTarget` cause exception 
[image]

**Additional context**
I have pysnmp 7.1.9 installed.

Ircama, 2024-11-04:
Did you follow the [requirements](https://github.com/Ircama/epson_print_conf/blob/main/requirements.txt) to setup the environment?

inspinet, 2024-11-04:
tried on linux box and worked first time so you're right, something got not right with windows environment

closed 2024-11-04 by inspinet