Ircama/epson_print_conf #39: ET-4800

pinpoint90, 2024-11-09:
Have the ET-4800 and added it as described in the readme. When running "Detect Access Keys" I end up with this error:

[ERROR] More than one pattern appears to be a serial number:
[ERROR] - found pattern "X8GT011188" at address 793
[ERROR] - found pattern "X8GT011188" at address 1604
[ERROR] Could not detect serial number.

I am able to analyze EEPROM values by pressing "Detect Configuration".
Looks like Reset button also work:

[NOTE] Current EEPROM values for Raw waste reset: 28: 19, 52: 209, 53: 15, 54: 94, 55: 94, 255: 94, 47: 0, 48: 147, 49: 11, 50: 89, 51: 0, 252: 110, 253: 0, 254: 0.
[NOTE] Current EEPROM values for Main waste: 48: 147, 49: 11, 47: 0.
[NOTE] Current EEPROM values for Borderless waste: 50: 89, 51: 0, 47: 0.
[NOTE] Current EEPROM values for Third waste: 252: 110, 253: 0, 254: 0.

I do not know the values for the ET-4800 and do not know how to check if these are correct. Do you think it is safe to go ahead and reset?

Ircama, 2024-11-09:
> [ERROR] - found pattern "X8GT011188" at address 793
> [ERROR] - found pattern "X8GT011188" at address 1604

Version 5.1 should not show an error when a set of serial number patterns is found, provided they are the same, as in the above case.

I have not tested that model, but I guess it is safe to go and it should be possible to reverse the operation in case of problems.

pinpoint90, 2024-11-09:
Ok thanks!
When running "Detect Configuration",  the following appears in a CMD window after a while:
Exception in Tkinter callback
Traceback (most recent call last):
  File "tkinter\__init__.py", line 1967, in __call__
  File "tkinter\__init__.py", line 861, in callit
  File "ui.py", line 2439, in <lambda>
  File "ui.py", line 2541, in detect_configuration
IndexError: list index out of range

Ircama, 2024-11-09:
> Traceback (most recent call last):
> File "tkinter_init_.py", line 1967, in call
> File "tkinter_init_.py", line 861, in callit
> File "ui.py", line 2439, in
> File "ui.py", line 2541, in detect_configuration
> IndexError: list index out of range

Strange, I simulated your pattern occurrence and I do not have this error; can you please double-check you have the latest code? Please reinstall the application. Can you repeat the operation and check that you exactly pasted all Traceback data? Which OS are you using?

pinpoint90, 2024-11-09:
Yes using the executable exe file v.5.1.0 on Wiundows 11 Pro, Tried to reinstall the app and the same happens again. 

Exception in Tkinter callback
Traceback (most recent call last):
  File "tkinter\__init__.py", line 1967, in __call__
  File "tkinter\__init__.py", line 861, in callit
  File "ui.py", line 2439, in <lambda>
  File "ui.py", line 2541, in detect_configuration
IndexError: list index out of range

Ircama, 2024-11-10:
Now the following issues should be solved:

> Exception in Tkinter callback......File "ui.py", line 2541, in detect_configuration...IndexError: list index out of range

> [INFO] Found data: {}.

Could you please check? Thanks

pinpoint90, 2024-11-10:
"Printer configuration": {
    "serial_name[0]": "X8GT011188",
    "serial_number[0]": range(793, 803),
    "serial_name[1]": "X8GT011188",
    "serial_number[1]": range(1604, 1614),
    "po_time": None,
    "Maintenance required level[0]": [54],
    "Maintenance required level[1]": [55],
    "Maintenance required level[2]": [255],
    "Maintenance required level[3]": [1680],
    "wifi_mac_address[0]": range(1920, 1926),
    "read_key": [74, 54],
    "write_key": b"Maribaya",
}

Thanks!
Seems to be working now. Does the "Maintenance required level" has anything to do with the ink waste pads and what does the values mean?

Ircama, 2024-11-10:
> Seems to be working now. Does the "Maintenance required level" has anything to do with the ink waste pads and what does the values mean?

They refer to the maximum percentage of waste level to require maintenance.

It is all ok now, thanks; I'll add the configuration of ET-4800.

closed 2024-11-10 by Ircama