Ircama/epson_print_conf #42: "Invalid Printer Model Selected" & similar - XP-7100

TutorialsOnDeck, 2024-11-14:
similar to #26 #29 and #35

**Printer model**

`XP-7100`

**Software configuration**

```
"XP-7100" + " configuration": {
    "read_key": [255, 254],
    "write_key": b"Leucojum",
    "main_waste": {"oids": [16, 17, 6], "divider": 84.5},
    "borderless_waste": {"oids": [18, 19, 6], "divider": 33.7},
    "raw_waste_reset": {16: 0, 17: 0, 6: 0, 52: 94, 20: 0, 21: 0, 18: 0, 19: 0, 53: 94, 493: 0},
    "stats": {
        "First TI received time": [9, 8],
        "Total print pass counter": [99, 98, 97, 96],
        "Total print page counter - front feed lower": [696, 695, 694, 693],
        "Total print page counter - front feed upper": [744, 743, 742, 741],
        "Total print page counter - rear": [748, 747, 746, 745],
        "Total print page counter - duplex": [752, 751, 750, 749],
        "Total print CD-R counter": [255, 254],
        "Total scan counter": [453, 452, 451, 450],
        "Total scan counter (ADF)": [457, 456, 455, 454],
        "Ink replacement counter - Black": [701],
        "Ink replacement counter % PB": [705],
        "Ink replacement counter - Cyan": [702],
        "Ink replacement counter - Magenta": [703],
        "Ink replacement counter - Yellow": [704],
        "Maintenance required level of 1st waste ink counter": [52],
        "Maintenance required level of 2nd waste ink counter": [53],
    },
    "serial_number": range(216, 226),
}
```

**Printer status**

```
"XP-7100": {
    "cartridge_information": [
        {"ink_color": [3391, "unknown color"], "ink_quantity": 55, "production_year": 2024, "production_month": 4, "data": "18041E0E000E1A0009", "manufacturer": "EPSON"},
        {"ink_color": [3402, "unknown color"], "ink_quantity": 78, "production_year": 2022, "production_month": 4, "data": "16040101380E03009E", "manufacturer": "EPSON"},
        {"ink_color": [3404, "unknown color"], "ink_quantity": 80, "production_year": 2021, "production_month": 7, "data": "150707072A0E0900A6", "manufacturer": "EPSON"},
        {"ink_color": [3403, "unknown color"], "ink_quantity": 78, "production_year": 2023, "production_month": 3, "data": "1703170E350E040003", "manufacturer": "EPSON"},
        {"ink_color": [3401, "unknown color"], "ink_quantity": 30, "production_year": 2023, "production_month": 1, "data": "1701120C260E030081", "manufacturer": "EPSON"},
    ],
    "cartridges": ["410XL", "410XL", "410XL", "410XL", "410XL"],
    "firmware_version": "YL25O7 25 Jul 2024",
    "printer_status": {
        "ready": True,
        "status": (4, "Idle (ready to print)"),
        "paper_path": b"\x01\x02",
        "unknown": [
            ("0xa", b"\x11\x00\x03"),
            ("0x1b", b"\xdd"),
            ("0x20", b"\x00ddq\x82fShhl\x80bRddh|d\\ggggggdddddddddddddddddddddddd"),
            ("0x24", b"\x00\x00"),
            ("0x39", b"\x00"),
            ("0x3f", b"\x01\x01\x02\x00\x01\x00\x00\x00\x00\x00\x01\x01%\n\x00\x00\x00\x00\x00"),
            ("0x45", b"\x18\x11\x0f\x0f\t"),
        ],
        "replace_cartridge": "00011111",
        "ink_level": [(11, 0, "Matte Black", "Black", 55), (3, 1, "Cyan", "Cyan", 78), (5, 3, "Yellow", "Yellow", 80), (4, 2, "Magenta", "Magenta", 78), (1, 0, "Black", "Black", 30)],
        "cancel_code": "No request",
        "jobname": "Not defined",
        "paper_count_normal": -1,
        "paper_count_page": -1,
        "paper_count_color": 6700,
        "paper_count_monochrome": 2872,
        "paper_count_blank": 241,
        "maintenance_box_1": "not full (0)",
        "maintenance_box_reset_count_1": 1,
        "maintenance_box_2": "not full (0)",
        "maintenance_box_reset_count_2": 0,
        "serial_number_info": "X5D5088515",
    },
    "serial_number": "?",
    "snmp_info": {
        "Model": "EPSON XP-7100 Series",
        "Epson Printer Name": "XP-7100 Series",
        "Model short": "XP-7100 Series",
        "Epson Personal Name": "XP-7100 Series",
        "EEPS2 firmware version": "EEPS2 Hard Ver.1.00 Firm Ver.0.72",
        "Epson Version number": "01.72.YL25O7.FY18",
        "Descr": "EPSON Built-in 11b/g/n & 10/100 Print Server",
        "UpTime": "03:57:09",
        "Name": "EPSONXP7100",
        "MAC Address": "38-9D-92-B0-2C-33",
        "Print input": "Auto sheet feeder",
        "Lang 1": "unknown",
        "Lang 2": "ESCPL2",
        "Lang 3": "BDC",
        "Lang 4": "D4",
        "Lang 5": "ESCPR2",
        "Emulation 1": "unknown",
        "Emulation 2": "ESC/P2",
        "Emulation 3": "BDC",
        "Emulation 4": "other",
        "Emulation 5": "other",
        "Total printed pages": "0",
        "IP Address": "",
        "IPP_URL": "http://10.0.0.6:631/ipp/print",
        "WiFi": "X21",
        "MAC Addr": "38-9D-92-B0-2C-33",
        "Epson device id": "MFG:EPSON;CMD:ESCPL2,BDC,D4,D4PX,ESCPR2,END4,GENEP;MDL:XP-7100 Series;CLS:PRINTER;DES:EPSON XP-7100 Series;CID:EpsonRGB;FID:FXN,DPA,WFA,ETA,AFA,DAA,WRA;RID:20;DDS:022500;ELG:0D00;SN:583544353038383515;",
        "Power Off Timer": "\x00ot:;\x0c",
    },
    "stats": {
        "First TI received time": "?",
        "Total print pass counter": None,
        "Total print page counter - front feed lower": None,
        "Total print page counter - front feed upper": None,
        "Total print page counter - rear": None,
        "Total print page counter - duplex": None,
        "Total print CD-R counter": None,
        "Total scan counter": None,
        "Total scan counter (ADF)": None,
        "Ink replacement counter - Black": None,
        "Ink replacement counter % PB": None,
        "Ink replacement counter - Cyan": None,
        "Ink replacement counter - Magenta": None,
        "Ink replacement counter - Yellow": None,
        "Maintenance required level of 1st waste ink counter": None,
        "Maintenance required level of 2nd waste ink counter": None,
    },
}
```

**Detect configuration**

GUI only shows this:
```
[INFO] Reading Printer SNMP values...
[INFO] Reading EEPROM values, please wait for some minutes...
[ERROR] Cannot read EEPROM values: invalid printer model selected.
```
By running it with -d and capturing the output
`https://pastebin.com/t046R5Lh` - over 12000 lines, expires in 2 weeks

**Describe the issue**

Ran both v5.1 UI.PY and 5.2.1 .exe (using WINE and a win10 VM) on my Epson XP-7100 to reset WIC and it reports my printer is not the correct model. Some functions of the software run without errors: Detect Printer (exe works not .py version), Printer Status, Printer Web Interface, (F2),(F3).

Non working functions (for me at least on XP-7100 on both 5.1 and 5.2.1)
- Get Poweroff Time
- TI Receive time
- Wifi MAC Address
- Printer Serial Number
- Detect Configuration
- Detect Access Keys
- Reset WIC

**Additional context**

I can run any commands on either linux or windows (VM) you want. Let me know how I can help. I've read through #25, #26, and #35

side note - Doing git clone shows the py version as 5.1 at the top of the GUI despite 5.2.1 being the latest. v0v

Ircama, 2024-11-16:
> side note - Doing git clone shows the py version as 5.1 at the top of the GUI despite 5.2.1 being the latest. v0v

Fixed with https://github.com/Ircama/epson_print_conf/commit/2f28901a0ae1b01aa5ac4ce9d5d88b13129e9070

-----------------

Please try following command and paste the output:

```
snmpget -v1 -d -c public 10.0.0.6 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.40.5.65.190.160.0.0
```

See also https://github.com/Ircama/epson_print_conf/issues/35#issuecomment-2451323220

TutorialsOnDeck, 2024-11-16:
read_key value was obtained from following the **issue report** instructions

```
**Software configuration**
In the UI, select the model from the drop-down list; press F2 in correspondence to your model. The status box will show the printer definition values. Select the status box, right button of the mouse, then copy all data and paste them here.
```
Steps taken when I first posted this issue:

1. I used "detect printer" which filled in XP-7100
2. I used F2
3. I copy pasted the results.

Retrying the above steps right now yielded a different value for the read_key [40, 5]. Tested both UI.py and the .exe - readouts are identical.

```
"XP-7100" + " configuration": {
    "read_key": [40, 5],
    "write_key": b"Leucojum",
    "main_waste": {"oids": [16, 17, 6], "divider": 84.5},
    "borderless_waste": {"oids": [18, 19, 6], "divider": 33.7},
    "raw_waste_reset": {16: 0, 17: 0, 6: 0, 52: 94, 20: 0, 21: 0, 18: 0, 19: 0, 53: 94, 493: 0},
    "stats": {
        "First TI received time": [9, 8],
        "Total print pass counter": [99, 98, 97, 96],
        "Total print page counter - front feed lower": [696, 695, 694, 693],
        "Total print page counter - front feed upper": [744, 743, 742, 741],
        "Total print page counter - rear": [748, 747, 746, 745],
        "Total print page counter - duplex": [752, 751, 750, 749],
        "Total print CD-R counter": [255, 254],
        "Total scan counter": [453, 452, 451, 450],
        "Total scan counter (ADF)": [457, 456, 455, 454],
        "Ink replacement counter - Black": [701],
        "Ink replacement counter % PB": [705],
        "Ink replacement counter - Cyan": [702],
        "Ink replacement counter - Magenta": [703],
        "Ink replacement counter - Yellow": [704],
        "Maintenance required level of 1st waste ink counter": [52],
        "Maintenance required level of 2nd waste ink counter": [53],
    },
    "serial_number": range(216, 226),
}
```
I don't see where you were getting [40, 5] from any logs I posted, Ctrl + F only shows you mentioning [40, 5]. Regardless, even with a read key value of [40, 5] All the the same errors persist.

From ui.py
```
[INFO] Detecting printers... (this might take a while)
[INFO] Found printer 'EPSON XP-7100 Series' at 10.0.0.6 (hostname: Unknown)
[ERROR] Cannot read EEPROM values for "Raw waste reset": invalid printer model selected.
[ERROR] time data '?' does not match format '%d %b %Y'
Traceback (most recent call last):
  File "/home/myusername/epson_print_conf/venv/bin/epson_print_conf/ui.py", line 1678, in get_ti_date
    date_string = datetime.strptime(
                  ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/_strptime.py", line 554, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/_strptime.py", line 333, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '?' does not match format '%d %b %Y'

[ERROR]: Cannot retrieve the printer serial number.
[INFO] Reading Printer SNMP values...
[INFO] Reading EEPROM values, please wait for some minutes...
[ERROR] Cannot read EEPROM values: invalid printer model selected.
```
If it would help, I can set you up a virtual machine (your OS choice) on my network for you to Teamviewer / Rustdesk remote to.

Ircama, 2024-11-17:
Please try again "Detect Access Keys"

Ircama, 2024-11-18:
Similarly to ET-2850 and L3250, there is no known way to read the EEPROM of XP-7100

Ircama, 2024-11-18:
> Similarly to ET-2850 and L3250, there is no known way to read the EEPROM of XP-7100
> There is! Works fine. It's named "leucojum". See <https://codeberg.org/atufi/reinkpy/>

The write key is

```
    "write_key": b"Leucojum",
```

(Capitalized)

Anyway how to read the EEPROM is unknown.

Ircama, 2024-11-18:
> I am still happy to assist in testing despite my problems being (possibly) resolved.

Can you try reading the EEPROM now, that is after being able to reset the waste counters (e.g., Get Printer Serial Number)?

dgerber, 2024-11-18:
Right, with capital L! But I don't get what you mean by "how to read the
EEPROM is unknown". It behaves like other models. I've had no special issue.

Ircama, 2024-11-19:
> Right, with capital L! But I don't get what you mean by "how to read the EEPROM is unknown". It behaves like other models. I've had no special issue.

Yes, it is supposed to behave like other models, but for some unknown reason it does not allow to read data from the EEPROM (at least the way I read data via SNMP).

In fact, before writing data to the EEPROM (`"write_key": b"Leucojum"`), I attempt to read data from it.

The read operation [fails](https://github.com/Ircama/epson_print_conf/issues/42#issuecomment-2480505394) supposing that the read key is `"read_key": [40, 5],`.

Is there a problem with this?

```
snmpget -v1 -d -c public 10.0.0.6 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.40.5.65.190.160.0.0
```

The message "*[ERROR] Could not detect read_key.*" means that there is no read sequence available for that printer (no [x, y] key works, for any number x and y between 0 and 255).

I cannot find a solution by now. Anyway, I have never tried to write data without reading them first.

TutorialsOnDeck, 2024-11-19:
> > I am still happy to assist in testing despite my problems being (possibly) resolved.
> 
> Can you try reading the EEPROM now, that is after being able to reset the waste counters (e.g., Get Printer Serial Number)?

Same Result: Cannot retrieve printer serial number

All other buttons that did work, work.
All buttons that did not work, still do not work.
[recap: Detect Printer (exe works not .py version), Printer Status, Printer Web Interface, (F2),(F3) - WORKS]

dgerber, 2024-11-19:
> ```
> snmpget -v1 -d -c public 10.0.0.6 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.40.5.65.190.160.0.0
> ```
This works on mine.

Ircama, 2024-11-20:
> ``` snmpget -v1 -d -c public 10.0.0.6 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.40.5.65.190.160.0.0 ```
> This works on mine.

Thanks, this helps.

@TutorialsOnDeck could you please check that snmpget ([Linux](https://manpages.ubuntu.com/manpages/jammy/man1/snmpget.1.html) or [Windows](https://netcologne.dl.sourceforge.net/project/net-snmp/net-snmp%20binaries/5.7-binaries/net-snmp-5.7.0-1.x86.exe?viasf=1)) works in the same environment where you run *epson_print_conf*?

dgerber, 2024-11-20:
>> ``` snmpget -v1 -d -c public 10.0.0.6 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.40.5.65.190.160.0.0 ```
>> This works on mine.
>
> Thanks, this helps.

Firmware version here is YL11K6. There is apprently no options related
to snmp in the user menus. So I've no clue what the issue on
@TutorialsOnDeck's device could be.

Ircama, 2024-11-21:
The string `||:41:NA;` likely represents an error, potentially indicating that the read key is not accepted. The issue might stem from a difference in the XP-7100 firmware versions: YL25O7 (25 Jul 2024) vs. YL11K6. I suspect that, in more recent firmware versions, the SNMP protocol for securing EEPROM access has been changed.

Ircama, 2025-01-04:
It appears that newer firmwares cannot be controlled via SNMP using the currently known method. You might use [reinkpy](https://codeberg.org/atufi/reinkpy/) via USB meanwhile.

io4x, 2025-01-04:
Thank you very much for your quick reply. The printer is not recognised via USB. It is recognised with other tools.

The printer is also displayed correctly under Printers in the system settings.

Asmodai88, 2025-01-05:
Hi, same problem here with a EPSON ET-2814 (74.54 doesn't work) I'm using the windows exe. and give me the error: Cannot read EEPROM values: invalid printer model selected.
F2 works, F3 works

io4x, 2025-01-08:
I was able to rectify the error by installing an older firmware.

Ircama, 2025-01-09:
> I was able to rectify the error by installing an older firmware.

Thanks for the info; which is the version of the working firmware? Did you use epson_print_conf to modify the printer EEPROM parameters?

io4x, 2025-01-09:
I found the file XP7100_YL11M7.zip - [XP7100_YL11M7.zip](https://github.com/user-attachments/files/18361999/XP7100_YL11M7.zip)

I ran the file EPFWUPD.EXE and followed all the steps until the firmware was successfully uploaded to the printer and then used the tool [reinkpy](https://codeberg.org/atufi/reinkpy/).

Ircama, 2026-09-09:
Your printer model is available in the Epson printer configuration database and can be imported into this program as described in the README: https://github.com/Ircama/epson_print_conf#how-to-import-an-external-printer-configuration-db

closed 2026-09-09 by Ircama

Ircama, 2026-09-22:
Version 8.0.0 introduces preliminary USB support.