Ircama/epson_print_conf #46: ET-7700

adacom72, 2024-12-22:
tryingto use the software with the ET-7700 - a lot of it works 
by choosing the ET-4800 config but not sure how accurate info given is

tried the reset waste setting but that will not work as it tells me wrong printer selected

windows 11 on a dell laptop - ethernet connection

Ircama, 2024-12-26:
ET-4800 is not the same as ET-7700.

Check these chapters in the documentation:

[How to import an external printer configuration DB](https://ircama.github.io/epson_print_conf/#how-to-import-an-external-printer-configuration-db)

[How to detect parameters of an unknown printer](https://ircama.github.io/epson_print_conf/#how-to-detect-parameters-of-an-unknown-printer)

adacom72, 2024-12-26:
thanks - i had found those links previously and used them - i got some info but have not worked out if its possible to then add the   printer thats found to the GUI - will look deeper after the holidays - ultimately i would like to reset the maintenance box if thats possible

closed 2025-02-10 by github-actions[bot]