Ircama/epson_print_conf #47: L18050 not manageable

dzidq, 2025-01-09:
I followed the instructions on importing external printer configuration and eventually managed to add a new profile for L18050. I successfully read the printer status, however I wasn't able to brute force the read_key (after several minutes I just got "_Could not detect read_key_." message). I guess this means that this particular model is not compatible and probably should be added to the "_Known incompatible models_" list. If you think my logs are of any use, I could post them here (although I don't think they're that interesting).

Ircama, 2025-01-10:
> ...I guess this means that this particular model is not compatible and probably should be added to the "_Known incompatible models_" list....

I'm afraid you are right. Thanks for your note.

dgerber, 2025-01-10:
I think this only means the EEPROM is not accessible *over SNMP*. It may
still be over USB.

But this model has a "maintenance box", with its own chip, so one
presumably needs dedicated hardware to reset the counter on it anyway.
Only the platen pad counters may be resettable via the main EEPROM.

Ircama, 2026-09-09:
Your printer model is available in the Epson printer configuration database and can be imported into this program as described in the README: https://github.com/Ircama/epson_print_conf#how-to-import-an-external-printer-configuration-db

closed 2026-09-09 by Ircama

Ircama, 2026-09-22:
Version 8.0.0 introduces preliminary USB support.