Ircama/epson_print_conf #51: Epson XP-2200 Series

PabloLpzz, 2025-02-07:
**Scenario**
Was able to get the read_key, but not the write_key:
[INFO] Starting the access key detection, please wait for many minutes...
[INFO] Detecting the read_key...
[INFO] Detected read_key: [75, 54].
[INFO] Detecting the serial number...
[INFO] Detected serial number "XB7R27****" at address 1604.
[NOTE] Current EEPROM value for the last byte of the serial number: 1613: 49.
[INFO] Detecting the write_key, do not power off the printer now...
[ERROR] Unable to detect the write key by validating against any of the known ones.

Maybe incompatible model?

Ircama, 2025-02-14:
> Maybe incompatible model?

Yes, I am afraid this model has a different write key than the known ones.

CiRIP, 2025-03-25:
Did you try `Lfokfsbo`?

Ircama, 2025-03-27:
@PabloLpzz to test @CiRIP's suggestion, you might follow these steps:

- Run the [*Epson Printer Configuration* GUI](https://github.com/Ircama/epson_print_conf/?tab=readme-ov-file#running-the-pre-built-gui-executable-code)
- Press F7 to get the *Debug Shell*
- Enter
  ```
  self.printer.PRINTER_CONFIG["XP-2200"] = {'read_key': [75, 54], 'write_key': b'Lfokfsbo', 'serial_number': range(1604, 1614)}
  ```
- If you then enter
  ```
  self.printer.write_key_list([75, 54])[0]
  ```
   you should get `b'Lfokfsbo'`
- Return to the *Epson Printer Configuration* GUI panel and press *"Detect Access Keys"*. It should detect the `read_key`, extract the serial number and use the last character of the serial number to validate the `write_key`.
- If no errors are reported in the output, report it and proceed by pressing *Detect Configuration*.
- Report its output

Check [How to detect parameters of an unknown printer](https://github.com/Ircama/epson_print_conf/?tab=readme-ov-file#how-to-detect-parameters-of-an-unknown-printer) for further information.

PabloLpzz, 2025-03-31:
Hi there, I tried and it didn't work:

>>>self.printer.PRINTER_CONFIG["XP-2200"] = {'read_key': [75, 54], 'write_key': b'Lfokfsbo', 'serial_number': range(1604, 1614)}
>>>self.printer.write_key_list([75, 54])[0]
b'Lfokfsbo'

[NOTE] Current EEPROM value for the last byte of the serial number: 1613: 49.
[INFO] Detecting the write_key, do not power off the printer now...
[ERROR] Unable to detect the write key by validating against any of the known ones.

Same output.

CiRIP, 2025-03-31:
Interesting as this key is taken directly from a decompiled XP-2200 firmware:

[image]

This is how it looks for an XP-600 for which we know the write key:

[image]

What firmware version are you on? Perhaps they changed the write key in the meantime?

CiRIP, 2025-03-31:
Oh I just realized that this tool might expect the keys in their rotated form. In this case, try `b'Kenjeran'` instead of `b'Lfokfsbo'`

PabloLpzz, 2025-03-31:
@CiRIP You were right!! Thank you so much!:
[INFO] Detecting the write_key, do not power off the printer now...
[INFO] Detected write_key: b'Kenjeran'
[ERROR] The selected write key b'Gazania*' is different from the detected one, which will be used to go on.
[INFO] Models with same read_key: ['XP-2200']
[INFO] Models with same write_key: ['XP-2200']
[INFO] Models with same access keys: ['XP-2200']
[INFO] Found data: {'read_key': [75, 54], 'serial_number': range(1604, 1614), 'write_key': b'Kenjeran'}.
[INFO] Detect operation completed.

Ircama, 2025-04-01:
Thanks @CiRIP , @PabloLpzz . This model could be possibly similar to the known XP-2100 one.

@PabloLpzz you might test the following at your risk:

- Run the [*Epson Printer Configuration* GUI](https://github.com/Ircama/epson_print_conf/?tab=readme-ov-file#running-the-pre-built-gui-executable-code)
- Press F7 to get the *Debug Shell*
- Enter
  ```
  self.printer.PRINTER_CONFIG["XP-2200"] = {
      "read_key": [75, 54],
      "write_key": b"Kenjeran",
      "main_waste": {"oids": [337, 338, 336], "divider": 69.0},
      "borderless_waste": {"oids": [339, 340, 336], "divider": 30.49},
      "raw_waste_reset": {336: 0, 337: 0, 338: 0, 339: 0, 340: 0, 341: 0, 343: 94, 342: 0, 344: 94, 28: 0},
      "stats": {
          "First TI received time": [9, 8],
          "Manual cleaning counter": [203],
          "Timer cleaning counter": [205],
          "Total print pass counter": [133, 132, 131, 130],
          "Total scan counter": [1843, 1842, 1841, 1840],
          "Total print page counter": [792, 791, 790, 789],
          "Ink replacement counter - Black": [554],
          "Ink replacement counter - Cyan": [555],
          "Ink replacement counter - Magenta": [556],
          "Ink replacement counter - Yellow": [557],
          "Maintenance required level of 1st waste ink counter": [343],
          "Maintenance required level of 2nd waste ink counter": [344],
          "Power off timer 1": [230, 229],
          "Power off timer 2": [231, 230],
          "Power off timer 3": [262, 261],
      },
      "serial_number": range(1604, 1614),
      "wifi_mac_address": range(1920, 1926),
  }; self.model_dropdown["values"] += ("XP-2200",)
  ```
- You can close the *Debug Shell*. Then return to the *Epson Printer Configuration* GUI, where the *XP-2200* model should appear as the last item in the *Printer Model* dropdown.

The *Printer Status* button is safe and you can test it.

You might be able to verify that the *TI Received Time* date matches when you first set up the printer.

If you try changing the *Power-Off Timer* in the printer driver configuration, you might discover which of the  Power off timer 1, 2 or 3 matches your changes.

Also, check if the other "stats" parameters produce reliable data.

Resetting the Waste Ink Levels might work, but testing it carries some risk. It is crucial to copy the log output and save it in a file to ensure the configuration can be reverted if something goes wrong.

The tank values may not exactly match the actual printer values, but this does not affect the reset function and is not harmful.

Murphydev585, 2025-05-03:
Please, where can I buy the xp2200 activation key

Ircama, 2025-05-04:
> Please, where can I buy the xp2200 activation key

Which activation key? There’s nothing to buy here.

andronedev, 2025-05-12:
> Thanks [@CiRIP](https://github.com/CiRIP) , [@PabloLpzz](https://github.com/PabloLpzz) . This model could be possibly similar to the known XP-2100 one.
>
> [@PabloLpzz](https://github.com/PabloLpzz) [...]

[image]

it doesn't seem to work for me

Ircama, 2025-05-12:
> it doesn't seem to work for me

Sorry, I corrected the [comment](https://github.com/Ircama/epson_print_conf/issues/51#issuecomment-2768017542) and added "Select a printer model and enter an IP address (or press the "Detect Printers" button)" before opening the Debug Console.

420pootang69, 2025-05-17:
I've just got an XP-2000 brand new in box. Is there something I can do to help?

CiRIP, 2025-05-19:
@420pootang69 Are you sure it's an XP-2000? That model does not exist on https://epson.sn...

RogueHoro, 2025-05-25:
Hi,

I’m using an **XP-2205** printer. I was able to connect to it and retrieve its status, but key detection doesn’t seem to be working. Here's the output I received:

```
[INFO] Detecting printers... (this might take a while)  
[INFO] Found printer 'EPSON XP-2200 Series' at 192.168.xxx.xxx (hostname: Unknown)  
[ERROR] Cannot retrieve the printer serial number.  
[INFO] Starting the access key detection, please wait for many minutes...  
[INFO] Detecting the read_key...  
[ERROR] Could not detect read_key.
```

**Firmware version:** `06.58.IU05P2`

Is this the end of the road, or is there something else I can try? Could the issue be related to the newer firmware?

I have an **XP-2150** an it works perfectly fine.

Thanks in advance for any insights.

Ircama, 2025-05-25:
>  I was able to connect to it and retrieve its status, but key detection doesn’t seem to be working.

Did you try applying the steps documented here (which are indeed referred to XP-2200)? https://github.com/Ircama/epson_print_conf/issues/51#issuecomment-2768017542

Ircama, 2025-05-28:
> I created a fork and added the XP-2200 entry to PRINTER_CONFIG, based on previous comments. Unfortunately, that didn’t seem to work.

According to your description, I'm afraid it's the recent firmware that disabled SNMP EEPROM access.

420pootang69, 2025-05-28:
I just want to chime in quick as I didn't respond earlier. It is an XP2200
I have.I didn't originally realise that this required WiFi access, which I
purposefully wouldn't connect as I'd be afraid of the unit getting online
somehow and getting firmware.
It's connected to my Linux box and networked through CUPS.

Ircama, 2026-09-09:
Your printer model is available in the Epson printer configuration database and can be imported into this program as described in the README: https://github.com/Ircama/epson_print_conf#how-to-import-an-external-printer-configuration-db

closed 2026-09-09 by Ircama