Ircama/epson_print_conf #54: Invalid argument for read_eeprom when reading values of EEPROM addresses with the CLI

bfayn, 2025-03-14:
The error `invalid argument for read_eeprom` is generated when using the option `-`R with the `epson_print_conf.py `command. The error occurs when the value read at an EEPROM address is greater or equal than 10 (`0x0A`).
The error ***does not*** happen with the GUI when reading value at the same EEPROM address.

For example reading the values for Main waste with the GUI shows in the Status :

`[NOTE] Current EEPROM values for Main waste: 24: 249,25: 23, 30:0. `

While running the command :

`python3 epson_print_conf.py -d -m ET-2650 -a 192.168.40.7 -R 24,25,30`

Displays:

```
EEPROM_DUMP read_eeprom:
  ADDRESS: 1.3.6.1.4.1.1248.1.2.2.44.1.1.2.1.124.124.7.0.16.8.65.190.160.24.0
  OID: 24=0x18
  TAG: OctetString
  RESPONSE: b'\x00@BDC PS\r\nEE:0018F9;\x0c'
invalid argument for read_eeprom
```

Error happens with software version **v5.3.6**

**Bug analysis**

The invalid argument is generated by the following line:

`2908:                             f"{int(val):#04x} = {val.rjust(3)}"`

Because `val` is already a representation of an hexadecimal value the `int(val):#04x` call generates an invalid argument error.

Proposed fix:

```
2908c2908
<                             f"{int(val):#04x} = {val.rjust(3)}"
---
>                             f"0x{val.rjust(2)} = {int(val,16)}"
```

Ircama, 2025-03-18:
Thanks for your appropriate analysis. Please feel free to open a pull request, or let me know if you'd prefer that I handle the fix.

bfayn, 2025-03-18:
I'm happy to push a commit and create a pull request but I don't have the correct rights (or I am doing something wrong):

```
$ git remote set-url origin git@github.com:Ircama/epson_print_conf.git
$ git push -u origin issue-54
ERROR: Permission to Ircama/epson_print_conf.git denied to bfayn.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

Ircama, 2025-03-19:
The correct approach should be:

- Fork this repository to your GitHub account using the GitHub UI. You should then get https://github.com/bfayn/epson_print_conf
- Clone the Fork to your dev platform

  ```
  git clone git@github.com:bfayn/epson_print_conf.git
  cd epson_print_conf
  ```
 
- Create a Branch

  ```
  git checkout -b issue-54
  ```

- Make Changes in the branch and Commit

  ```
  git commit -a -m "Fix for issue #54"
  ```

- Push to your Fork (not this)

  ```
  git push -u origin issue-54
  ```

- Using the GitHub UI, create a Pull Request from your issue-54 branch to this repository.

Check also [GitHub’s documentation](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests) on forking and pull requests.

bfayn, 2025-03-19:
Many thanks for your help Ircama!

commit 5fa1db18 (2025-03-20): Fix for issue #54 (#55)

Fixes #54, thanks
closed 2025-03-20 by Ircama