Ircama/epson_print_conf #60: Collab request

buterrip69, 2025-04-27:
Hi!

First, thank you for your awesome work with Epson printers via SNMP.

We have started a new open-source project focusing on Epson ET-2870 (and similar models) to reset waste ink counters using SNMP and USB communication.

Repo: https://github.com/buterrip69/epson-et2870-waste-reset

We would love your feedback or collaboration, especially around writable OIDs discovery and testing methodology.

Would you be interested in helping or advising us?

Thanks again for your work!

**Scenario**
Clearly describe that you followed the instructions in the README and specifically that you carefully completed the steps described in the [Example to integrate new printers](https://github.com/Ircama/epson_print_conf/?tab=readme-ov-file#example-to-integrate-new-printers). Report the issue including steps to reproduce the behavior and a short description of what you expected to happen.

**Printer model**
Provide the exact printer model.

**Printer status**
Press "Printer status" in the UI. The status box will show a tree view with printer data. Select the status box, right button of the mouse, then copy all data and paste them here.

**Detect Access Keys**
Press "Detect Access Keys". After the operation is completed, the status box will show a tree view with printer data. Select the status box, right button of the mouse, then copy all data and paste them here.

**Detect configuration**
Press "Detect configuration". After the operation is completed, the status box will show a tree view with printer data. Select the status box, right button of the mouse, then copy all data and paste them here.

**Additional context**
Feel free to add any other context about the problem here.

Ircama, 2025-04-27:
I reviewed your code — it appears to rely on __libusb__ as a backend, which I'm not particularly inclined to use.

Unfortunately, I have limited time and cannot directly test USB functionality. However, feel free to submit a pull request to this repository, adapting the USB-reset.py logic to behave similarly to [epson_print_conf.py](https://github.com/Ircama/epson_print_conf/blob/main/epson_print_conf.py); I will review your contribution.

buterrip69, 2025-04-27:
Hey there, waste of time on my end too. I tried ReinkPy and it works for 2870 on models 2813-2855. Thank you btw.

Ircama, 2026-09-22:
Version 8.0.0 introduces preliminary USB support.