Ircama/epson_print_conf #64: CRITICAL:root:snmp_mib invalid address

ANSB, 2025-05-14:
Hi,
I'm new in Python (but start coding in ASM in 1987 and play a lot with C and PHP for years!).
First congrat for the job and the code. I installed Python on my Mac and run ui after installing yaml and so on.

I think I'll have some time to check and correct the code but here are the problems:
Detect printer -> is able to find my 2 Epson but unable to get the name (Unknown found at 192.168.0.149 (hotsname: Unknown). When  set this IP on browser, I get the internal web site of the printer so, I see the two IP are the ips of the printers.

In fact, there is a lot of small errors, all displaying same message in the debugger.
CRITICAL:root:snmp_mib invalid address: Please call .create() to construct UdpTransportTarget object

I'll check for that.

Another point: where to you find the data for printer? I have a L355 which is in the list but also a L396 which is not in the list... Where can I find the L396 data?

Thanks very much, and once again, "Good Job"
PL

ANSB, 2025-05-14:
Sorry, a file was missing!

closed 2025-05-14 by ANSB

Ircama, 2025-05-15:
> In fact, there is a lot of small errors, all displaying same message in the debugger. CRITICAL:root:snmp_mib invalid address: Please call .create() to construct UdpTransportTarget object

The error you're seeing:

```
CRITICAL:root:snmp_mib invalid address: Please call .create() to construct UdpTransportTarget object
```

indicates improper usage of the `pysnmp` library (possibly installed via `pip install pysnmp`), which currently points to the **lextudio fork**. This version is **asynchronous-only** and requires explicit `.create()` calls for transport initialization — making it unsuitable for simple, sequential SNMP v1 workflows.

For the use case in this repo (synchronous SNMP v1 queries to a single host) the [original backend](https://github.com/Ircama/epson_print_conf/blob/52ce58adda7ce4ced7c21501561e26e9576a9f49/requirements.txt#L3) from https://github.com/etingof/pysnmp was adopted.

You should uninstall the lextudio version:

```bash
pip uninstall pysnmp
```

and install the library conforming to the [specifications](https://github.com/Ircama/epson_print_conf?tab=readme-ov-file#installation):

```bash
pip install git+https://github.com/etingof/pysnmp.git@master#egg=pysnmp
```

I have [also tested](https://github.com/Ircama/epson_print_conf/tree/pysnmplib) `pysnmplib` (`pip install pysnmplib`) which is a newer backend, but it’s slightly slower than the original `pysnmp` implementation.

---------------------

> Another point: where to you find the data for printer? I have a L355 which is in the list but also a L396 which is not in the list... Where can I find the L396 data?

All specifications in this repo are directly collected from printers, or sourced from public [referenced sites](https://github.com/Ircama/epson_print_conf/tree/pysnmplib?tab=readme-ov-file#references) and have been tested either directly or indirectly. Please refer also to the [documentation](https://github.com/Ircama/epson_print_conf/tree/pysnmplib?tab=readme-ov-file#how-to-import-an-external-printer-configuration-db).

ANSB, 2025-05-15:
In fact I choose to uninstall "all" and then install "all" again and that's OK. So maybe an install problem. For my L396 after using Detect Access Keys I discoverd that L396 = L3060. As I'm using a Mac, I'll continue testing but after that I'll see how to create an "exe" but for the Mac and send it to you. :)