Ircama/epson_print_conf #72: L5290 not found in the list.

lejunosorio, 2025-07-15:
epson L5290 is detected (via IP address), but throw an error: Printer Model Unknown

**Printer model**
Provide the exact printer model.

**Printer status**
Press "Printer status" in the UI. The status box will show a tree view with printer data. Select the status box, right button of the mouse, then copy all data and paste them here.

**Detect Access Keys**
Press "Detect Access Keys". After the operation is completed, the status box will show a tree view with printer data. Select the status box, right button of the mouse, then copy all data and paste them here.

**Detect configuration**
Press "Detect configuration". After the operation is completed, the status box will show a tree view with printer data. Select the status box, right button of the mouse, then copy all data and paste them here.

**Additional context**
Feel free to add any other context about the problem here.

Ircama, 2025-07-21:
Please check documentation: https://github.com/Ircama/epson_print_conf?tab=readme-ov-file#how-to-import-an-external-printer-configuration-db

Ircama, 2025-08-25:
I'm afraid your printer firmware disabled the SNMP management of the printer EEPROM.

closed 2025-10-21 by github-actions[bot]