Ircama/epson_print_conf #73: program indentifys the printer L3251 (L3250), but cant find the model

HelaGhost, 2025-07-21:
Hi,  (i'm not that tech savvy, so that might be why i have some trouble)

I'm trying to reset the ink waste counter in Epson L3251 printer
Installed the program, launched it through cmd.exe 
The model of the printer doesn't seem to on the list, even with "Detect printers" the program does not register as a possibility 

[image]

Can you help me with it, or my printer is not compatible with this program?
Thank you in advance

Ircama, 2025-07-21:
Please check documentation:

- https://github.com/Ircama/epson_print_conf?tab=readme-ov-file#how-to-import-an-external-printer-configuration-db
- https://github.com/Ircama/epson_print_conf?tab=readme-ov-file#known-incompatible-models

HelaGhost, 2025-07-21:
Sorry, i'm an idiot... 
I got to this part

[image]