Ircama/epson_print_conf #74: XP455

QuentinBisson, 2025-07-22:
**Printer model**
XP455

**Software configuration**
In the UI, select the model from the drop-down list; press F2 in correspondence to your model. The status box will show the printer definition values. Select the status box, right button of the mouse, then copy all data and paste them here.

**Printer status**
Press "Printer status". The status box will show a tree view with printer data. Select the status box, right button of the mouse, then copy all data and paste them here.

**Detect configuration**
Press "Detect configuration". After the operation is completed, the status box will show a tree view with printer data. Select the status box, right button of the mouse, then copy all data and paste them here.

**Describe the issue**
All commands are failing for this one

**Additional context**
Optionally add any other context about the problem here.

Ircama, 2025-07-23:
Please check documentation: https://github.com/Ircama/epson_print_conf?tab=readme-ov-file#how-to-import-an-external-printer-configuration-db

QuentinBisson, 2025-07-23:
Oh my bad, thank you :)